reilleya/openMotor #34: Pressure transients

reilleya, 2019-04-17:
Do some research into loosening the assumption that pressure is at steady state. It would be cool to (optionally) simulate things like startup pressure from a known igniter or burst disk, and also to estimate how pressure drops off after all propellant has been consumed.

adam-aitoumez, 2019-05-29:
Steady-state assumption is generally alright, but limits the modeling of burst disk or chamber filling effects as you mentioned, as well as analyzing motor sensitivity to things like rapid increase in burning area or decrease in throat area. A better method is to use

<img src="http://latex.codecogs.com/svg.latex?\frac{dP_c}{dt} = \frac{R_g T_f}{V_c}\Big( r_b A_b (\rho_p - \rho_g) - \frac{P_c A_t}{C^*} \Big)" border="0"/>

where _R<sub>g</sub>_ is the combustion products specific gas constant,  _T<sub>f</sub>_ is the combustion temperature,  _V<sub>c</sub>_ is chamber free volume, and  _&rho;<sub>g</sub>_ is combustion gas density, all of which either products of grain geometry or pressure and propellant-dependent constants that come from an equilibrium combustion solver.

reilleya, 2019-05-29:
Thanks for doing the research for me! Are you interested in implementing this? I will if you don't, but it would be nice to have additional collaborators.

PS: Congrats on the space shot! I have been following RPL for a few years now and was very excited to see that it finally worked out.

adam-aitoumez, 2019-05-29:
I am interested - I see this going hand-in-hand with implementing lookup of tabular thermochemistry data as alluded to in https://github.com/reilleya/openMotor/issues/31#issuecomment-484758401, so I can take a crack at doing both of those. I personally only use CEA so I think my parser/interpolated lookup may be different from what's needed for ProPEP parsing, but it shouldn't be too hard to make it extensible to that case.

I'll dig into how y'all currently implement the steady-state pressure calculation, may have to play around with how the motor is initialized to get the time-unsteady equation working. It would also be good to implement a variable time step solver akin to MATLAB's `ode45` in conjunction with this, because in even the largest amateur motors chamber filling takes very little time (<0.1 sec for USCRPL's Traveler 3-class motor). Blow down is a little longer, but regardless a tiny time step is necessary to accurately capture transients whereas a larger time step can be used for steady-state operation.

Also, thank you!! It's been a deeply exciting several years to be a part of the team, and it's always quite cool to hear that folks have been following our journey and all of its ups and downs.

reilleya, 2024-12-17:
This feels like a 2.0 feature, like really only worth doing if the simulation backend is going to be made much more robust. For now, it just slowed down simulations significantly but produced nearly identical results.

closed 2024-12-17 by reilleya