reilleya/openMotor #35: Configurable alerts

reilleya, 2019-04-17:
Each motor file (or preferences?) should have a set of associated limits for pressure, mass flux, etc that generate alerts if they are passed.

reilleya, 2019-05-18:
@tuxxi Any chance you will have time to work on this soon? If not, I'm going to push out a v0.2.0 soon because all of the other features are almost done. This (along with the refactor) will be the first things we tackle in v0.3.0. No huge rush, but I want to get the mac ui fixes out ASAP.

tuxxi, 2019-05-21:
Hey, sorry I have not had much free time these past few days. I wanted to fix a few bugs in the erosive branch and PR it, but go ahead and put out 0.2.0 and we can work on this after.

reilleya, 2019-05-21:
No problem. Thanks for contributing your time!

closed 2019-06-14 by reilleya