reilleya/openMotor #40: replace loadUi() calls with pre-compiled pyuic5 files

tuxxi, 2019-04-19:
We can use https://pypi.org/project/pyqt-distutils/0.5.2/ to help manage .ui files so they are statically compiled by setup.py. This is apparently more "pythonic" than using `loadUi()` (which is also quite slow, it essentially runs pyuic each time at runtime.

After the .ui files are compiled, we can then `import` the compiled python source code as a normal file. This enables better inspection in addition to a speed boost.

Thoughts?

reilleya, 2019-04-19:
I liked using separate .ui files early on in development because it removed a step from the process of testing changes to the UI. Now that things are a bit more stable and the UI isn't changing as much, I think going for speed makes more sense. Startup does feel a bit slow and this is likely the culprit. Would you mind taking this one?

tuxxi, 2019-04-25:
Another startup time note I found while investigating how to properly profile the appication's startup time: 

I noticed that when running the win64 release, it took around 10 seconds to start!
Supposedly, creating the windows executable from PyInstaller in "one file" mode means that it has to unpack all the data directories before it can run -- especially the Qt framework is very large, so this would explain the huge startup time on the release build.

https://stackoverflow.com/questions/9469932/app-created-with-pyinstaller-has-a-slow-startup

reilleya, 2019-04-26:
I had experimented with both and decided that the single file was ok for now, but I will definitely be looking into an installer for a future version. Single file mode is definitely slow to start (probably unacceptably so) but the combination of Qt, matplotlib, and scikit makes the single directory method really messy. On a test version I made a while back there were 139 files/folders in the directory!

tuxxi, 2019-04-27:
After profiling the ui refactor branch, it appears to only be faster by about 15-25% - i.e it takes about ~0.35 seconds on average on my branch vs ~0.45 seconds on master. 

However, IDEs now give better inspection and autcomplete results (for UI elements before, it simply didn't work since the object's `__dict__` was loaded dynamically).
Another benefit is that (IMHO) the code is far more clear, as anything dealing with ui elements is referenced via `self.ui.{pushButton/Table/etc..}`.

closed 2019-04-29 by reilleya