reilleya/openMotor #47: Improved nozzle calculations

reilleya, 2019-04-30:
Change how thrust is calculated. Calculations should take into account details like convergent/divergent section angles and throat length to reduce dependence on the "efficiency" that thrust is currently scaled by.

reilleya, 2019-05-10:
http://www.rasaero.com/TECH_SUMMARY.htm shows how to relate thrust coefficient to nozzle geometry.

tuxxi, 2019-05-10:
Check out my old code here for some other ideas. including some other departures from idea performance such as skin friction:
https://github.com/tuxxi/OpenBurn/blob/master/openburn/core/internalballistics.py#L165

BurnSim's "efficiency" is called "two-phase flow efficiency" in the literature afaict.

reilleya, 2019-05-10:
Thanks, that is exactly what I was looking for!

closed 2019-07-09 by reilleya