reilleya/openMotor #55: Grain editior

benrussell11, 2019-05-07:
I really like the designer and how the Face etc are updated in real time as you change the various design parameters.

I've been doing some testing of the grain editor with the latest build.
- End Burner - Doesn't display the tab showing Face/Regression/Area Graph. Should it?

- Errors- If you have any errors in  the grain design, the errors are caught and reported when your "run" the sim. I believe it would be more user friendly if grain errors were displayed when the clicks on the Apply button.

- Add Grain button - When you click on "Add grain" it looks like the grain is added to the grid then the grain editor is displayed. If I cancel w/o editing or realize I picked the wrong grain type and wanted to cancel and not add the grain, the grain is still listed in the grid. Is it possible to first display the grain editor and then add the grain to grid when the user selects "Apply". Cancel would not add the grain.

Ben

reilleya, 2019-05-08:
- It would always just show a circle, so I didn't think it added any value.
- Good idea. I could add a small table below the preview (or maybe on a tab in the preview?) that lists the errors. Which would you prefer to see?
- This would be a bit difficult to do with how the code is set up right now. It isn't hard in theory but would require rewriting a decent amount of code. I agree that it is annoying, but I think it isn't a high priority for now.

reilleya, 2019-05-08:
- The area graph would be a flat line, so not super helpful. The regression graph only shows how the face regresses, and it would never have any contours on it because endburners regress only in length. 
- Can do! It would be nice to see errors/warnings while editing geometry. I think I will put this one in for 0.3.0.
- The issue is that the motor editor applies its changes to the currently selected grain in the table when the 'apply' button is pressed. I would have to add a special case for it to work on grains that are not yet in the table. Could you make a new issue for this? In general, I like to stick to a single topic of discussion per issue. 
- I can bump up the number of digits that are displayed in the grain/nozzle table to include as much precision as the user has entered. It would be great if you could make a new issue for this one as well.

We'll use this issue for discussion about the grain editor alerts table. Thanks for putting your time into testing the software and giving suggestions. I'm really excited for where it'll be in a few months!

-Andrew

reilleya, 2019-06-15:
This is now on staging

benrussell11, 2019-06-15:
Fixed ... I really like the errors being caught during design. I only tested Bates. Will try and test some of the other geometries.

closed 2019-06-30 by reilleya