reilleya/openMotor #60: Document motorlib

reilleya, 2019-05-09:
Add docstrings to motorlib functions.

quigleyj97, 2019-06-22:
Do we want to adopt a particular documentation generator? Looks like most functions in motorlib have docstrings but they're not readable by those generators (beyond just treating them as remarks).

This [StackOverflow q](https://stackoverflow.com/questions/3898572/what-is-the-standard-python-docstring-format) goes over a few different standards.

I can work on moving the existing docstrings to something like sphinx, which is fairly bog-standard, but I wanted to touch base before starting in case you had any thoughts.

Somewhat related (but probably deserving of it's own issue) is how to generate and distribute docs. GitHub pages would probably work well for this.

reilleya, 2019-06-24:
I didn't develop a strong preference while looking at the suggested formats, so we can go with reST if it is common or you prefer it. Feel free to post an issue about documentation distribution.

closed 2019-06-30 by reilleya