reilleya/openMotor #62: Export .eng file

benrussell11, 2019-05-09:
Motor Diameter - default units is inches. Suggest changing it to "mm" as all of our case id is in mm.

Ben
[image]

reilleya, 2019-05-10:
The unit for this field matches the user's current unit for length, but this isn't great because of the way the rocketry tends to mix units. I don't intend to make this field use any units other than the user's current setting, so I will mark this as "wontfix" for now. In the future, I may build a system to let users maintain a library of their hardware (nozzles, cases, mandrels, etc) and then this dialog would be updated to allow you to pick a specific case from a list instead of needing to enter all dimensions every time.

closed 2019-05-10 by reilleya

benrussell11, 2019-05-10:
Ok … but a diameter isn’t a length … how about adding a Diameter units to Preferences.

reilleya, 2019-05-10:
Diameter is not length, but they have the same units. The way unit conversion works in oM is that each field has units associated with it, and its value is scaled automatically based on the user's selected unit for that type (length, mass, density, etc). The system only handles one preference per unit type, and it would take some work to make it use different conversions for different fields of the same unit type.

benrussell11, 2019-05-10:
Ok … that is going to take some getting use to … let’s see what other users have to say after the .2 release.

 

Ben