reilleya/openMotor #64: Burnsim import - differences

benrussell11, 2019-05-09:
Import Burnsim file #22 with pressure data.bsx and saved as #22 BS Import.ric. The following are noted differences.

- Nozzle efficiency - BS - 1.0 OM - .01
- Expansion Ratio - BS 1.02 OM - 1.0 - 2 decimal places???
- Motor Designation - BS J227 OM - I137
- Impulse - BS 770 OM 429.274
- Delivered ISP - BS 156 OM 87.16
- Max Pc - BS 599.6 OM 689.971 - From test stand - Max pc = 1151, Ave Pc 739. Ave Pc is used derive A&N.
Ave Pc - BS NA OM - 618
Peak Max flux - BS 0.364 OM 0.313

Propellant is "2019 IREC Bule 1". User was warned that the propellant didn't exist in OM and added.

[#22 BS Import.ric.txt](https://github.com/reilleya/openMotor/files/3162814/22.BS.Import.ric.txt)
[#22 with pressure data.bsx.txt](https://github.com/reilleya/openMotor/files/3162815/22.with.pressure.data.bsx.txt)

tuxxi, 2019-05-10:
Ben,

Thanks for the report. It seems BurnSim usually stores nozzle efficiency as a raw percentage: e.g the 85% default is written as `NozzleEfficiency="85"`. _However_, files that have nozzle efficiency as 1.0 (very rare, its usually 93-96% even for very large motors fwiw), it stores it as `NozzleEfficiency="1"`!

For the expansion ratio, I'm not sure why BS is reporting `1.02`, in your file it says: `Nozzle ThroatDia="0.34375" ExitDia="0.34375"`, i.e exit == throat, so expansion ratio will be 1.0. 

Everything else is a function of the incorrect nozzle efficiency.

benrussell11, 2019-05-10:
Hi,

 

Ah nice catch …

 

I went back and set the BS nozzle efficiency to 85, expansion ratio to 1 and OM efficiency to .85.

 

What is interesting is I can change the nozzle efficiency in BS to anything from 0 to what ever, the sim output is always the same. It’s as if it is using 85% all of the time.

 

OM is now a lot closer to BS. BS J227 vs OM J286, Total NS BS 770 vs OM 893. OM sim is closer to the test data, if you looked at the test data that included with BS file.

 

Still curious as to why OM has calculated a higher total NS?

 

I’ll be importing more examples over the next couple of day.

 

BTW – you are correct in not importing gain type Tablet as it was done for a customer. A end burner in BS is a Bates with a “0” core.

 

Ben

reilleya, 2019-05-10:
Efficiency mean different things in oM and BS, and I actually like BS's method better and plan to make thrust in oM more similar in v0.3.0. In BurnSim, a motor will deliver the propellant's isp* if it either has an expansion ratio of 1.0 or an efficiency of near 0. If the motor has >1 expansion ratio, the ISP will be whatever the isp* was with another term added on for the nozzle expansion's contribution to thrust. This other term is what efficiency scales.

In oM, the thrust is just scaled by the efficiency. I plan to implement the equations here http://www.rasaero.com/TECH_SUMMARY.htm to make a reasonable guess of thrust coefficient based on nozzle geometry. The user will still be free to set "efficiency" to whatever they wish, but the goal is that the default value would be close to 100% because the thrust coefficient is already being adjusted by the program.

closed 2019-07-13 by reilleya