reilleya/openMotor #71: Simulation of small endburners crashes

Ross04, 2019-06-01:
This engine crashes openmotor 0.2.0. Because it is possible to make this engine it should be fixed. 
[openmotor_crash_data_zipped.zip](https://github.com/reilleya/openMotor/files/3244085/openmotor_crash_data_zipped.zip)
Alternatively, you could prevent engines causing this crash from being created. .ric file found several layers deep in the zip file.

reilleya, 2019-06-02:
The problem is that this motor has less than 'A' impulse, so the function that assigns the motor designation fails. I will make it able to assign classes below 'A' for the next version. Thanks!

reilleya, 2019-06-02:
After some more investigation, it seems that the reason this motor causes a crash is because it doesn't make enough pressure to produce thrust. It only has a Kn of 4, which is far too low for the propellant to pressurize the motor case. Typical APCP formulations run in the 100-400 Kn range. You can fix this motor by making the nozzle throat smaller or the propellant area larger, but you will likely still run into the error I mentioned first as the motor is so small. I will add a new condition to stop the simulation when pressure drops too low for the motor to be doing anything.

benrussell11, 2019-06-13:
Fixed in Stage. Really ...

closed 2019-06-13 by reilleya