reilleya/openMotor #72: Comparison CSV export Vs Motor Stats UI display

benrussell11, 2019-06-04:
I exported to CSV the attached motor file.  Below are the differences between the spreadsheet and the Motor Stat UI display. 
[98 with smoke grain 5-5.ric.txt](https://github.com/reilleya/openMotor/files/3254203/98.with.smoke.grain.5-5.ric.txt)

  | Motor Stats | Export
-- | -- | --
Peak KN | 243.053 | 243.0521
Ave Pc | 467.073 | 503.0016
Max Pc | 524.898 | 564.6619
BT | 4.29 | 4.11
Peak Flux G:7 | 1.424 | 1.53042

Shouldn't they be the same?

Ben

reilleya, 2019-06-04:
It is possible that you ran a simulation, tweaked the motor, exported the CSV, and ran the simulation again, in that order? Both the motor stats and the CSV include results from the last simulation, but exporting a CSV does not run a new simulation. If you run a simulation, change the motor, and export a CSV without rerunning the simulation, the data exported will be from the outdated simulation. I'm guessing this was the problem, because when I open your motor, run a simulation, and export a CSV, the stats and CSV match perfectly. I suppose the CSV exporter (and exporters in general) should automatically run a simulation of the current motor to avoid this kind of situation.

benrussell11, 2019-06-05:
You are 100% right … I changed the density and ran the sim. I then compared it to an older version of the spreadsheet. I should have exported after the changes and then done the comparison.

 

Ben

reilleya, 2019-06-05:
No problem, it is a bit unclear exactly how exporting works right now. Do you think it is better to have it run a new simulation whenever you export, or should it disable exporting unless you have run a simulation since you last changed the motor?

benrussell11, 2019-06-05:
I think it’s best to run a sim before exporting. One less step for the user and everything is in sync.

closed 2019-06-11 by reilleya