reilleya/openMotor #74: Tools - Design - Neutral Bates Geometry

benrussell11, 2019-06-05:
I've been working with the above to design 98 mm bates grains. I really like it. One thing I noticed was that the TPL created didn't match what I entered. I think I understand why. Most of the time it was 1 grain short for the starting KN. Like most hobbyist, I have cases, nozzles and mandrels of various sizes.. Would it be possible to have the system create the number of grains and length if I gave it the following:

1. Propellant Length and if needed grain spacer length
2. Propellant diameter
3. Core diameter
4. Nozzle throat
5. Starting KN

I'd also like to suggest automatically running the sim after Apply.

reilleya, 2019-06-13:
The issue with the tool as it stands right now is that the grains are all added at the length that makes them exactly neutral. The tool is usually stuck with some odd space in the liner that is shorter than this ideal grain length, and it currently does nothing with it. The two options are to shorten all of the grains slightly and add an additional grain, or to lengthen them a bit. Either choice will produce a motor that is still approximately neutral. Any ideas about which you would prefer? I will make it so the tool can accept an optional core diameter, but using both options 4 and 5 would result in an over-constrained system. People have been asking for a system that tracks that user's available nozzles and mandrels a lot on TRF, so I will start sketching up what that might look like and think about adding it to the roadmap for v0.4.0 or v0.5.0.

benrussell11, 2019-06-13:
I think I would prefer to lengthen the grains to match the TPL. Adding a optional core size is a good idea. I would be able to choose either a starting KN or a core size. Right? Understand that the grains will not necessarily be bates neutral. When I run the sims, I’ve noticed that in almost all cases, the Port/Throat ratio and Peak Mass Flux for the bottom grain are almost not usable. That is Port/Throat is almost always less than 2 and Peak Max flux is usually > 2 for given starting KN. I can play with the length and starting KN and usually get a Port/Throat > 2 and a Peak Max Flux < 2. 

 

Looking ahead a bit. I definitely like the idea of being able to have the system know what hardware I have. Would we be able to do something like this:

 

*	Select a formula
*	Enter TPL
*	Select Casting tube. In the first iteration, enter the ID of the casting tube.
*	Select the Nozzle which would define the throat size or enter it.
*	Select the mandrel size or enter it.
*	Specify a Port/Throat range/constrain
*	Specify a Peak Max Flux range/constrain

Run the sim which met the above constrains. Not sure that is even possible. If you don’t specify either the Port/Throat or Peak Max Flux, the system would figure out the best combination which not necessarily would be a neural bates grain.

 

Even further down the road, support multiple grain geometries.

 

Not sure this is even possible. 

 

Ben

BlazingAngel665, 2019-08-14:
Responding to Andrew's comment on June 20, perhaps we could create a system that tracks nozzles and cases, and adds them to the motor list, through an interface like the propellant editor. If we wanted to go the extra mile we could add Loki and RCS default nozzles.
Tracking available mandrels is a more interesting challenge. Perhaps for BATES grains it could have a dropdown when you click in the core diameter box that lets you select a 'saved' core diameter or type a new one?