reilleya/openMotor #76: Exported eng file will not load in EngEdit

benrussell11, 2019-06-09:
The "WillNotLoad.eng" file exported from the included ric file will not load in the EngEdit. This last time entry is negative and not 0. I believe the last time entry must be 0. Also negative newtons doesn't makes to me. To get it to load, change the last time entry to 4.32 0

I have not tried to load the unchanged eng file in Rocksim or openRocket.

[WillNotLoad.eng.txt](https://github.com/reilleya/openMotor/files/3269695/WillNotLoad.eng.txt)
[98 with smoke grain 5-5.ric.txt](https://github.com/reilleya/openMotor/files/3269697/98.with.smoke.grain.5-5.ric.txt)

reilleya, 2019-06-11:
It looks like the smoke grain continues to burn after the propellant burns out, as expected. It produces low enough chamber pressure that the solver is unable to calculate thrust from it. I will make it so thrust cannot  go below 0 in the next version. Thanks!

reilleya, 2019-06-13:
This should be fixed on staging.

benrussell11, 2019-06-13:
Fixed in stage.

commit 95922059 (2019-06-12): Fixed #76, #71, and #78
closed 2019-06-13 by reilleya