reilleya/openMotor #77: Build an import/export manager

reilleya, 2019-06-11:
As the number of importers and exporters grows, it will become annoying to have them all hardcoded. Build a system similar to the tool manager that handles the flow of information in and out of the importers/exporters that also populates the menu.

coop44483, 2019-07-15:
Is it possible to import only the propellant info from a BurnSim file.  I am the author of ProPep 3 which can create a BurnSim file, but it does not include any grain information.  I attempted to import one of my files into OpenMotor but nothing seems to happen.  
It would be nice to be able to import just the 7 values needed for a new propellant since some of my user can be a little computer challenged ;).
BTW  Great Program.  Chuck Rogers and I have talked about doing this but I looks like you may have beaten us to the punch!

Dave Cooper

reilleya, 2019-07-16:
Hey Dave, I hadn't considered a propellant-only import, but that seems like a good idea that will require almost no effort. I will make an issue ticket for the feature. Thanks! I have used some resources written by Chuck during development and would love to know if he has any ideas for the program, especially about how to simulate or predict erosive burning.

coop44483, 2019-07-16:
How can Chuck contact you?  I know he will not use github.  Can you send me a private message with your email address?  Or you could just contact Chuck using his aol account.  (RASAreo has it.)

I will contact him to give him a heads up.

Coop

 

This email has been checked for viruses by AVG.
https://www.avg.com

reilleya, 2019-07-17:
Thanks Coop, I will reach out through to him through his email.

reilleya, 2019-08-02:
@coop44483 The latest commit to staging adds support for importing burnsim files that contain only propellant data. I've tested it with a file generated by ProPEP 3 and it seems to work, but let me know if you notice any issues.

coop44483, 2019-08-03:
Wow That was fast.  I am currently not trying to keep up with OpenMotor since I have about 4 other projects going.  If you would give me a heads up when the new version is out and I will try running some of my data files for more testing.  Thanks.

Was Chuck able to help?

Coop

 

https://www.avg.com

closed 2019-08-03 by reilleya