reilleya/openMotor #78: File - Export - Image

benrussell11, 2019-06-11:
I exported Mass Flow and Flux graph images. I exported to Excel and compared the Excel graphs to the exported. Noted the following differences:

1. Vertical scale is not right in the exported graphs.
2. Exported Mass Flux doesn't look anything like the Excel graph.
3. Exported graphs look to be almost the same only the legend is different.

[image]
[image]
[98 with smoke grain 5-5.ric.txt](https://github.com/reilleya/openMotor/files/3277850/98.with.smoke.grain.5-5.ric.txt)
[Contest 5-5 with smoke.xlsx](https://github.com/reilleya/openMotor/files/3277851/Contest.5-5.with.smoke.xlsx)

reilleya, 2019-06-11:
Good catch! This is a pretty major bug. It looks like the scale of the axes is set by the first image you export, and then the axes aren't cleared properly if you export another one. I'm assuming you had exported a thrust curve first and then made these plots. Either way, this bug can be fixed with a single line of code and you will definitely see it fixed in 0.3.0 or maybe a 0.2.1 release if 0.3.0 takes too long. Thanks!

reilleya, 2019-06-12:
In case you want to play around with them, the fix to this issue and a few others are up in the `staging` branch. Most of the changes I have made for v0.3.0 so far have been related to restructuring the code for the UI, so it would be helpful to get a lot of general testing done on this branch. To change over to staging, navigate to your local cloned version of the program, and run: 
```
git fetch
git checkout staging
pip install -r requirements.txt
python setup.py build_ui
python main.py
```
Let me know if you have any problems.

benrussell11, 2019-06-12:
Actually I opened openMotor, loaded the RIC file, ran the sim and then did the exports to image. It looked like the “Y” axis came from the sim and not an expert of thrust.

benrussell11, 2019-06-12:
Followed the direction below for staging and will continue to testing using staging.

 

I should be able to do some testing this week. Next week is off the Spaceport America Cup (IREC).

reilleya, 2019-06-12:
Is it fixed on staging? Sounds good.

benrussell11, 2019-06-13:
Not working in Stage.

 

Try the following:

*	Open openMotor
*	Select a formula
*	Tools – Design – Bates
*	Enter info and Apply
*	Ctrl-R to run. If you haven’t selected a formula, you’ll receive the error message “Motor must have a propellant set”, which is correct
*	Sim is displayed
*	Tools – Design – Bates
*	Enter info – sure would be nice if the system saved what was enter previously
*	Apply
*	Ctrl-R run – you’ll receive the error message “Motor must have propellant set”. It was set from the last sim.
*	To get rid of the message

*	Select a different propellant
*	Select the one you want to sim.
*	CTRL-Run – Sim runs w/o error.

reilleya, 2019-06-13:
This thread is about image exporting. I have not yet pushed the fix for the BATES tool propellant issue.

benrussell11, 2019-06-13:
Sorry about that … that’s what I get for not paying attention to the subject.

 

The exported image/graph is correct for Stage.

 

I did get the following error which didn’t seem to affect the export of the graph/image.

 

Traceback (most recent call last):

  File "C:\openMotor\uilib\imageExport.py", line 51, in exportImage

    plt.clear()

AttributeError: module 'matplotlib.pyplot' has no attribute 'clear'

 

After saving the file and returning to the main screen, the app crashes on the first click of a mouse. Can’t tell if the above error is what causes the crash or not.

reilleya, 2019-06-13:
Ouch, this one was just a typo. Should be fixed if you pull the latest version.

reilleya, 2019-06-13:
Any issue that I put the "staging" label on should have its fix on staging. Make a comment on the thread if the original issue was fixed, and then I will close the issue. This is about what we have been doing, but adding the labels gives you a (hopefully short) todo list of stuff to check whenever you have time.

benrussell11, 2019-06-13:
Ok will do.

benrussell11, 2019-06-13:
The graphing has been fixed. However the app aborts after saving to the exported file. No error message except this one.

  File "C:\openMotor\uilib\imageExport.py", line 51, in exportImage
    plt.clear()
AttributeError: module 'matplotlib.pyplot' has no attribute 'clear'

reilleya, 2019-06-13:
Are you sure you pulled it down after 12:50 PM EST? I pushed a fix that works on my end.

benrussell11, 2019-06-14:
I just pulled down the latest version in stage. Clicking the mouse after the image has been saved, still aborts after the image has been created.

Microsoft Windows [Version 10.0.17134.829]
(c) 2018 Microsoft Corporation. All rights reserved.

C:\Users\ben>cd \openMotor

C:\openMotor>stage

C:\openMotor>git fetch
remote: Enumerating objects: 71, done.
remote: Counting objects: 100% (71/71), done.
remote: Compressing objects: 100% (22/22), done.
remote: Total 71 (delta 49), reused 59 (delta 49), pack-reused 0
Unpacking objects: 100% (71/71), done.
From https://github.com/reilleya/openMotor
 * [new branch]      motorlib_cleanup -> origin/motorlib_cleanup

C:\openMotor>git checkout staging
Already on 'staging'
Your branch is behind 'origin/staging' by 3 commits, and can be fast-forwarded.
  (use "git pull" to update your local branch)

C:\openMotor>pip install -r requirements.txt
Requirement already satisfied: appdirs==1.4.3 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 1)) (1.4.3)
Requirement already satisfied: cycler==0.10.0 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 2)) (0.10.0)
Requirement already satisfied: decorator==4.4.0 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 3)) (4.4.0)
Requirement already satisfied: docopt==0.6.2 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 4)) (0.6.2)
Requirement already satisfied: ezdxf==0.9 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 5)) (0.9)
Requirement already satisfied: imageio==2.5.0 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 6)) (2.5.0)
Requirement already satisfied: kiwisolver==1.0.1 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 7)) (1.0.1)
Requirement already satisfied: matplotlib==3.0.3 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 8)) (3.0.3)
Requirement already satisfied: networkx==2.3 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 9)) (2.3)
Requirement already satisfied: numpy==1.16.2 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 10)) (1.16.2)
Requirement already satisfied: Pillow==6.0.0 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 11)) (6.0.0)
Requirement already satisfied: pyparsing==2.4.0 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 12)) (2.4.0)
Requirement already satisfied: pyqt-distutils==0.7.3 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 13)) (0.7.3)
Requirement already satisfied: PyQt5==5.12.1 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 14)) (5.12.1)
Requirement already satisfied: PyQt5-sip==4.19.15 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 15)) (4.19.15)
Requirement already satisfied: python-dateutil==2.8.0 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 16)) (2.8.0)
Requirement already satisfied: PyWavelets==1.0.3 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 17)) (1.0.3)
Requirement already satisfied: PyYAML==5.1 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 18)) (5.1)
Requirement already satisfied: scikit-fmm==2019.1.30 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 19)) (2019.1.30)
Requirement already satisfied: scikit-image==0.15.0 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 20)) (0.15.0)
Requirement already satisfied: scipy==1.2.1 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 21)) (1.2.1)
Requirement already satisfied: six==1.12.0 in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from -r requirements.txt (line 22)) (1.12.0)
Requirement already satisfied: setuptools in c:\users\ben\appdata\local\programs\python\python37-32\lib\site-packages (from kiwisolver==1.0.1->-r requirements.txt (line 7)) (40.8.0)
WARNING: You are using pip version 19.1, however version 19.1.1 is available.
You should consider upgrading via the 'python -m pip install --upgrade pip' command.

C:\openMotor>python setup.py build_ui
running build_ui
skipping C:\openMotor\uilib/views/forms\AboutDialog.ui, up to date
skipping C:\openMotor\uilib/views/forms\CSVExporter.ui, up to date
skipping C:\openMotor\uilib/views/forms\EngExporter.ui, up to date
skipping C:\openMotor\uilib/views/forms\GrainPreview.ui, up to date
skipping C:\openMotor\uilib/views/forms\ImageExporter.ui, up to date
skipping C:\openMotor\uilib/views/forms\MainWindow.ui, up to date
skipping C:\openMotor\uilib/views/forms\Preferences.ui, up to date
[... cut ...]
C:\openMotor\uilib\fileIO.py:34: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  fileData = yaml.load(readLocation)
QLayout: Attempting to add QLayout "" to MotorEditor "motorEditor", which already has a layout
Traceback (most recent call last):
  File "C:\openMotor\uilib\imageExport.py", line 51, in exportImage
    plt.clear()
AttributeError: module 'matplotlib.pyplot' has no attribute 'clear'

reilleya, 2019-06-14:
Ah, the instructions I posted were for getting your copy to be on the staging branch. Now that it is on the staging branch, what you have to do to pull down the latest version is: 
```
git pull
python setup.py build_ui
```
Sorry for the confusion!

benrussell11, 2019-06-14:
No problem ... I really need to go thru Git tutorials ...

I've modified my "Stage" bat file to a pull and build. I'll run it before I start testing.

The problem with aborting after saving of the image has been fixed.

closed 2019-06-14 by reilleya