reilleya/openMotor #80: Tools - Design - Bates

benrussell11, 2019-06-12:
A propellant has been selected before entering bates grain info. After applying, the grain grid and nozzle are populated. Run the sim and get the error message that a propellant has been selected. Have to select a different propellant and then the one I want to use. After selecting, the sim runs.

reilleya, 2019-06-13:
This should be fixed on staging.

benrussell11, 2019-06-13:
I updated from Stage ... This is not fixed. Doing a 2nd design still pops up the error "Motor must have a propellant set". There is one set in the drop down. You have to select another propellant and then the one you want to use before re-running the sim.

benrussell11, 2019-06-14:
This have been fixed. Stage version 0.3.0

closed 2019-06-14 by reilleya