reilleya/openMotor #82: File - Export - Image

benrussell11, 2019-06-14:
If you haven't run a sim before exporting an image, the image dialog still pops up with the error message as part of the dialog. You can still select what you want to plot. The Ok button is greyed out and the Cancel button is active.

Suggest displaying the simulation not having been run error as a separate dialog and not display the Export Image dialog.  If you want to display the error as part of the Export image dialog, the error should be highlighted in "Red" and bolded.  Also one should not be able to select anything. Perhaps hide the X-Axis and Y-Axis frames.

reilleya, 2019-07-13:
This will be fixed when I build the import/export managers for #77 .

reilleya, 2019-07-30:
These changes are now on staging. Specifically, I made it so a popup shows up instead of the relevant exporter if you haven't run a simulation.

closed 2019-08-03 by reilleya