reilleya/openMotor #84: Negative and tabular propellant data

mariohm1311, 2019-06-16:
For most propellants, burn rate cannot be modelled correctly by Vieille's Law. Prime examples of these non-ideal behaviors are potassium nitrate/sugar formulations such as KNSB, which forms a plateau.

[image]
[Credit to Richard Nakka.](http://www.nakka-rocketry.net/sorb.html)

It would be a very nice addition to have the ability to characterize any propellant curve as a piecewise power law (as defined by Vieille's Law). This includes negative exponents, which currently make openMotor crash.

reilleya, 2019-06-17:
Good suggestion! I saw this plot while perusing Nakka at some point and decided I would add this functionality if someone asked for it. I'll add this in the release after the one I am currently developing unless you have some Python experience and enough time to hash it out yourself. I would excitedly accept a PR with this functionality!

reilleya, 2019-09-01:
This feature is now on staging if you want to check it out @mariohm1311

closed 2019-09-26 by reilleya