reilleya/openMotor #99: Documentation Distribution

quigleyj97, 2019-07-13:
Follow-up from https://github.com/reilleya/openMotor/issues/60#issuecomment-504802695

What mechanism do we want to use to distribute docs?

A few options we could go with:

 - [ReadTheDocs.io](https://docs.readthedocs.io/en/stable/intro/getting-started-with-sphinx.html)
  - [Github Pages](https://pages.github.com/)
  - Self-hosted
  - ...etc.

I think RTD is most appropriate for our use case, but I don't have any experience operationalizing it. I've done a few gh-pages projects before, and while it's very neat as an end-product it gets very nasty to manage (uses git, so it pollutes your git history and doesn't support building-from-source except from markdown).

reilleya, 2019-07-14:
I'm fully in favor of anything that keeps the repo history clean. Looks like RTD is the best option for that. Want me to mark this as part of v0.4.0?

quigleyj97, 2019-07-14:
Sounds good to me. I just submitted #100 with the Sphinx docs, if you'd like to check it out.

quigleyj97, 2019-07-21:
I've setup something on RTD: https://openmotor.readthedocs.io/en/latest/index.html

If you create a RTD account I can make you maintainer. You'll also need to setup the GitHub integration since I don't have permissions for that.

reilleya, 2019-07-23:
Thanks for doing this! I set up an account under the username "reilleya".

ashwani-rathee, 2020-06-07:
@reilleya @quigleyj97 Hey!!I would love to contribute by helping with maintaining the documentations..I don't have much experience but I have contributed to bokeh's documentation in past(rather insignificant though)..