reilleya/openMotor #101: HiDPI display errors on Windows 10

quigleyj97, 2019-07-14:
On hidpi displays, the app clips a few UI elements and fails to properly lay out others. I've pasted screenshots below detailing some of the problems I found.

Normally Qt should handle this but it looks like it isn't (or maybe I have a few wires crossed about the extent Qt is involved- I ran into similar issues on other Qt apps)

This is on a Surface Pro 4, which has a ~12" ~4k display at 200% UI scaling.

There is a workaround for this, but it's rather onerous for an end user: you need to override the scaling by right-clicking on the app in the taskbar, then right clicking the executable in the menu that pops up, "Properties" > "Compatibility" > "Change high DPI settings" > "Override application scaling" and set it to "System" or "System (enhanced)".

Clipping of UI elements:
[image]

Alerts dialog is too small, editors are not visible until you expand them:
[image]
[image]

Propellant editor is too cramped to use by default:
[image]

(Separately I can't get to the resize grip on this window- not sure if this is related).

reilleya, 2019-07-14:
Ouch, that's pretty bad! It looks like Qt is allowing the text to be resized by the system scale factor but not giving it room in the UI. I didn't use any fixed coordinates for UI elements or anything, so I don't think that is the problem. The Qt docs seem to have some solutions here: https://doc.qt.io/qt-5/highdpi.html. I can't really test this myself as I don't have anything particularly high DPI, so would you like to look into the solutions on the doc page?

quigleyj97, 2019-07-20:
Looking into this now. Setting `QT_AUTO_SCREEN_SCALE_FACTOR=1` fixes most resolution issues, the only major issues so far have been window size (at the minimum size it's almost as big as my screen) and the legend position on the simulation graph.

[image]

PR #103 (2019-07-24): Change: Set EnableHighDpiScaling flag for high DPI support
This is handled at the OS layer by OSX and at the server layer by Wayland, whereas X11 and Windows place the responsibility of scaling on the app. Qt includes a flag to enable high-dpi support trivially, though it causes the app to be unwieldy because it's minimum size is still quite large.

I took a stab at a few other avenues, since it should "just work", but it seems to be pretty deep into the weeds with how Qt layouts work and I don't have enough experience with Qt to resolve them right now.

I don't have access to an OSX device, s [... cut]
commit dff7e956 (2019-07-24): Change: Set EnableHighDpiScaling flag for high DPI support

This is handled at the OS layer by OSX and at the server layer by Wayland, whereas X11 and Windows place the responsibility of scaling on the app. Qt includes a flag to enable high-dpi support trivially, though it causes the app to be unwieldly large.

Fixes #101
closed 2019-08-03 by reilleya