reilleya/openMotor #108: Finocyl Regression Error

BlazingAngel665, 2019-07-27:
[image]
The area graph of this finocyl is producing odd results. Note that following the regression curves shows weird behavior near the top fin at about 75% of the y height of the grain.
[image]
Clean File with glitchy behavior attached.
Preliminary investigation shows this behavior is exists on 5 fin, 8 fin, 11 fin, 13 fin, and 19 fin geometries. The bug persists over different core and grain diameters, as well as over different fin thicknesses and lengths.
[Finocyl_Glitch.zip](https://github.com/reilleya/openMotor/files/3437806/Finocyl_Glitch.zip)

reilleya, 2019-07-27:
Ouch! Was this on master or staging?

BlazingAngel665, 2019-07-27:
Staging

mariohm1311, 2019-08-01:
This is reproducible in 0.3.0. Additionally, decreasing time step worsens oscillations. 

[image]

[image]

mariohm1311, 2019-08-01:
This isn't just limited to Finocyl: star, moon burner, X core, C and D grains have all been tested and found to suffer from the same bug. 

[image]

[image]

[image]

[image]

[image]

mariohm1311, 2019-08-01:
Increasing grain map dimension reduces the magnitude of the oscillations, so this narrows the problem down to the FMM solver.

Map dimension = 750
[image]

Map dimension = 1500
[image]

reilleya, 2019-08-01:
Mario is correct, as this issue appears for FMM grains, which includes everything but end burners, BATES, and now rod and tube grains. The issue is that grain mass is calculated by taking the grain's regression image (where pixel value corresponds to regression depth) and summing up the area that exceeds the grain's current regression depth. This leads to grain mass vs regression depth having a lot of steps in it because of the discrete nature of the image. Mass flow is the derivative of that steppy function, so it ends up very chaotic. I tried to do some smoothing to the values (see commit 4cf9419b17f59d532ae1c87b90b6a0f9fd3d15a3) but it seems to have not worked if this still shows up on staging. I will revisit this and try something else to smooth it out. Does anyone have any recommendations for how to smooth out the step function produced by the regression image?

PS: @mariohm1311 I plan to tackle tabular propellant data next. Let me know if you have any suggestions for what you want the UI to look like.

reilleya, 2019-08-01:
As a side note, the issue with the contours shown on the regression plot is annoying but purely a visual defect.

mariohm1311, 2019-08-01:
I would suggest running traditional upsampling methods for the regression map such as bilinear or bicubic interpolation, since those are cheap to compute, and see if the result is any better. If not, something slightly more involved such as pixel-shift super-resolution techniques can be applied (a naive implementation isn't that complex). 

@reilleya Regarding tabular propellant data, if anything just make sure you accept negative exponents. Burnsim has this feature, yet doesn't accept negative exponents, so :/. Once you have some prototype, please post some pictures on the relevant issue and ping me, so I can give you some suggestions. For now, I'll let you do what's easier for you.

reilleya, 2019-08-03:
Well, it still isn't perfect, but I have a decent solution for this pushed to staging. Let me know if you find any configurations that are awful. I will define awful as "resembling any of the examples above other than the last one, which is only kind of bad".

closed 2019-08-25 by reilleya