reilleya/openMotor #116: Deriving Thrust from Pressure data

benrussell11, 2019-08-14:
Hi Andrew,

I've built and have been using a pressure data logger, which we have been using with our test stand to derive A&N.

My next step is to fly the pressure logger and log real time flight pressure. I would like to be able to convert the logged pressure to "thrust" and compare flight pressure to simed pressure. I can compare logged pressure vs simed pressure by exporting the simed pressure. Easy...

One way is to have openMotor import the logged pressure data and derive thrust using the motor design file (nozzle, grain geo etc). That would be an enhancement.

Is there a simple formula for determining thrust from the logged pressure? If there is, could you provide me with the formula? I can do the conversion within the data logger or export time and pressure to an Excel compatible CSV file and do the conversion in Excel.

Thanks,

Ben

BlazingAngel665, 2019-08-14:
With the pressure and nozzle geometry it's pretty straightforward to get thrust.
The equation using only nozzle parameters is:
`thrust = A* P_c * sqrt((2gamma**2/(gamma-1)) (2/(gamma+1))**((gamma+1)/(gamma-1))*(1-(P_e/P_c)**(gamma-1/gamma))) + (P_c-P_e)*A_e` or rather,
[image]

benrussell11, 2019-08-15:
Hi Charlie,

 

Thanks a million. Simple… well … is this the same formula openMotor is using?

 

Terms for the equation below:

*	A* is the area of the throat
*	P_c is the pressure at a point in time
*	What is gamma? What is Y?
*	P_e is atmospheric pressure – 14.1??? psi
*	What is A_e?
*	Is this the equation openMotor is using?

 

What units should I be working in? Pressure is in PSI and would like to have force in Newtons.

 

Are you and Andrew attending this month’s Potter launch?

 

Again thanks for all of the help.

 

Ben

 

 <https://user-images.githubusercontent.com/4712964/63063380-69403e80-beca-11e9-9e31-125c1aeaaf2b.png>

BlazingAngel665, 2019-08-16:
This wouldn't be how OpenMotor would calculate thrust, to my knowledge, since OM can calculate mass flow directly, and calculates pressure off of KN.

The terms you provided are correct.
Gamma (a greekified lower-case 'y') is the specific heat ratio of your propellant, or Cp/Cv.
P_e is the exit pressure, the pressure of your exhaust at the exit plane of the nozzle. This is computed from the expansion ratio.
A_e is the area of the nozzle at the exit plane (radius of nozzle exit squared times pi)
You should be working entirely in metric units. Pa, N, kg, m, etc.
I think Andrew and I will be at URRG.

benrussell11, 2019-08-19:
Hi Charlie,

 

A couple of follow up questions,

 

*	Isn’t what is calculated by the sqrt part of the equation the Thrust Coefficient? As near as I can tell, openMotor doesn’t calculate it. Does it make sense or could OM calculate the CF and include it as one of the export items?
*	In ProPep 3 and for a given formula, Chamber CP/CV is calculated. Is this the same as specific heat ratio?
*	Exit pressure P_e – How do I calculate p_e given the expansion ratio? How do I calculate the expansion ratio? OM calculates an expansion ratio, can I use it?
*	For a straight thru throat, does A_e = A*?

 

I’ve received the 1600 pressure transducers and will be working on data logger software this week.

 

Thanks for all of your help.

 

Ben

BlazingAngel665, 2019-08-20:
Thrust coefficient is the term in square root with a nozzle area normalization term. Open Motor could calculate Cf. Andrew and I have been discussing improvement to the nozzle modelling and display process, this might be part of that.

The Chamber Cp/Cv is the same thing as specific heat ratio. There is a minor possibility for inaccuracy as the fraction of exhaust products (and therefore specific heat ratio) changes as the gas cools in the nozzle, but it's still mostly accurate.

Expansion ratio is Ae/A* and is typically in the range of 10 to 25 for amateur motors. The expansion ratio calculated by OM is fine to use. To calculate P_e use the equations here:
https://www.grc.nasa.gov/WWW/K-12/airplane/rktthsum.html
Use Expansion Ratio to get the exit mach. Then use the exit mach number to calculate the exit pressure.

If your nozzle is literally a hole in a flat plate, then A_e=A_t.

reilleya, 2019-08-20:
The software already calculates Cf and I planned to display it on the nozzle tab of the results widget once that exists.

benrussell11, 2019-08-20:
If I understand correctly, that should simplify my equation. Will it be included as an option for export?

 

Any feeling for when it will be added to the nozzle tab?

 

Ben

benrussell11, 2019-08-20:
Will do and thanks.

 

Ben

 

https://www.grc.nasa.gov/WWW/K-12/airplane/rktthsum.html

closed 2019-09-27 by reilleya