reilleya/openMotor #117: Grain by Grain Web

BlazingAngel665, 2019-08-20:
Add a row to the 'Grains' tab of the results widget which shows the thickness of propellant which burns to reach burnout.

reilleya, 2019-09-01:
This is on staging now if you want to check it out. Be sure to build the UI after pulling.

benrussell11, 2019-09-02:
1 - Just updated to “Stage”. Doing a simulation on any of my design files never completes. However, I can “cancel” out of the simulation.

 

2 – Open a design, Click on Propellant Editor, click on Edit w/o having selected a propellant, the system crashes.

 

3 – Propellant editor suggestions

*	Double click to edit a propellent
*	Selected a propellant and then click on Edit – the system crashes
*	Main screen – If a propellant has been selected, clicking on the Propellant Editor button displays the propellant editor with the selected propellant loaded. For me at least, if I have a propellant selected on the main screen and select the its editor, I want to “go” to that propellant.

 

Here’s the log from my sessions. Hope this helps.

 

Microsoft Windows [Version 10.0.17763.678]

(c) 2018 Microsoft Corporation. All rights reserved.

 

C:\Users\ben>cd \openmotor

 

C:\openMotor>stage

 

C:\openMotor>git pull

Already up to date.

 

C:\openMotor>python setup.py build_ui

running build_ui

skipping C:\openMotor\uilib/views/forms\AboutDialog.ui, up to date

skipping C:\openMotor\uilib/views/forms\CSVExporter.ui, up to date

skipping C:\openMotor\uilib/views/forms\EngExporter.ui, up to date

skipping C:\openMotor\uilib/views/forms\GrainPreview.ui, up to date

skipping C:\openMotor\uilib/views/forms\ImageExporter.ui, up to date

skipping C:\openMotor\uilib/views/forms\MainWindow.ui, up to date

skipping C:\openMotor\uilib/views/forms\NozzlePreview.ui, up to date

skipping C:\openMotor\uilib/views/forms\Preferences.ui, up to date

skipping C:\openMotor\uilib/views/forms\PropellantPreview.ui, up to date

skipping C:\openMotor\uilib/views/forms\PropMenu.ui, up to date

skipping C:\openMotor\uilib/views/forms\ResultsWidget.ui, up to date

skipping C:\openMotor\uilib/views/forms\SimulatingDialog.ui, up to date

skipping C:\openMotor\uilib/views/forms\SimulationAlertsDialog.ui, up to date

skipping C:\openMotor\uilib/views/forms\TabularEditor.ui, up to date

 

--- Open design file and run sim

 

C:\openMotor>python main.py

C:\openMotor\uilib\fileIO.py:37: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.

  fileData = yaml.load(readLocation)

QLayout: Attempting to add QLayout "" to MotorEditor "motorEditor", which already has a layout

Exception in thread Thread-1:

Traceback (most recent call last):

  File "C:\Users\ben\AppData\Local\Programs\Python\Python37-32\lib\threading.py", line 917, in _bootstrap_inner

    self.run()

  File "C:\Users\ben\AppData\Local\Programs\Python\Python37-32\lib\threading.py", line 865, in run

    self._target(*self._args, **self._kwargs)

  File "C:\openMotor\uilib\simulationManager.py", line 44, in _simThread

    simRes = self.motor.runSimulation(self.updateProgressBar)

  File "C:\openMotor\motorlib\motor.py", line 183, in runSimulation

    simRes.channels['pressure'].addData(self.calcIdealPressure(perGrainReg, igniterPres, None, burnoutWebThres))

  File "C:\openMotor\motorlib\motor.py", line 86, in calcIdealPressure

    ballA, ballN, gamma, temp, molarMass = self.propellant.getCombustionProperties(lastPressure)

  File "C:\openMotor\motorlib\propellant.py", line 60, in getCombustionProperties

    return closest['a'], closest['n'], closest['k'], closest['t'], closest['m']

KeyError: 'a'

 

--- Open design file, click on Propellant Editor, select a propellant, click on Edit

 

C:\openMotor>python main.py

C:\openMotor\uilib\fileIO.py:37: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.

  fileData = yaml.load(readLocation)

QLayout: Attempting to add QLayout "" to MotorEditor "motorEditor", which already has a layout

Traceback (most recent call last):

  File "C:\openMotor\uilib\widgets\propellantMenu.py", line 76, in editProp

    self.ui.propEditor.loadProperties(prop)

  File "C:\openMotor\uilib\widgets\propellantEditor.py", line 26, in loadProperties

    super().loadProperties(obj)

  File "C:\openMotor\uilib\widgets\collectionEditor.py", line 79, in loadProperties

    self.propertyUpdate()

  File "C:\openMotor\uilib\widgets\propellantEditor.py", line 23, in propertyUpdate

    self.propellantPreview.loadPropellant(previewProp)

  File "C:\openMotor\uilib\widgets\propellantPreviewWidget.py", line 29, in loadPropellant

    minPres = int(propellant.getMinimumValidPressure()) + 1 # Add 1 Pa to avoid crashing on burnrate for 0 Pa

  File "C:\openMotor\motorlib\propellant.py", line 64, in getMinimumValidPressure

    return min([tab['minPressure'] for tab in self.getProperty('tabs')])

ValueError: min() arg is an empty sequence

 

C:\openMotor>

reilleya, 2019-09-02:
I think the cause is that you ran a previous staging version before this and it migrated your motors to the old version of 0.4.0 before I added tabular propellants. Do you have 0.3.0 versions of your stuff around still? Those should open properly. You should also be able to make new propellants without an issue. Let me know if you still have problems. A user going directly from 0.3.0 to the real release version of 0.4.0 would not have this problem, but unfortunately there is no great way to avoid incompatibilities between staging versions. Basically, there are functions that convert motors, preferences, and propellants between versions, but they only work between actual releases (0.3.0 and 0.4.0) not staging versions.

benrussell11, 2019-09-02:
The first thing I did this morning was update from stage.  After doing so, I closed and came back and tried a few things which I documented below. So it looks like my “propellant” file has been clobbered and I’ll need to reenter all of propellants because I don’t have a backup of propellants.yaml. I’ll add backing up propellants.yaml and preferences.yaml as part of updating to the latest version of yaml. Might want to consider having the program automatedly backup these files for the just in case there are future incompatibilities.

 

Yes I can create and edit new propellants. I wasn’t prepared for the new dialog, but really like what you are doing. I’m recreating my Tiger Tail propellant and have everything except for Exhaust Molar Mass. From where do I get the Exhaust Molar Mass? I’m using ProPep 3 for Specific Heat Ratio, Combustion Temp. Calculated Characteristic Velocity isn’t right. See attached screen shots and the TT ProPep3 file.

 

Defaults for the new propellant screen – shouldn’t all of the defaults be 0.00 or just blank? For instance burn rate exponent defaults to -1, specific heat ratio to 1.0001, exhaust molar mass 0.000001.

 

Ben

 

<Document>
  <FileVersion>1</FileVersion>
  <PropellantData>
    <ChamberPressure>1000</ChamberPressure>
    <ExhaustPressure>14.7</ExhaustPressure>
    <Temperature>298</Temperature>
    <Boost>False</Boost>
    <PropellantName>Tiger Tail</PropellantName>
    <Comments />
    <Ingredient>
      <Name>ALUMINUM (PURE CRYSTALINE)</Name>
      <Wt>11.500</Wt>
    </Ingredient>
    <Ingredient>
      <Name>AMMONIUM PERCHLORATE</Name>
      <Wt>445.600</Wt>
    </Ingredient>
    <Ingredient>
      <Name>HTPB (R-45HT)</Name>
      <Wt>103.500</Wt>
    </Ingredient>
    <Ingredient>
      <Name>MDI (PAPI 94)</Name>
      <Wt>14.700</Wt>
    </Ingredient>
    <BurnRateCoefficient>0</BurnRateCoefficient>
    <BurnRateExponent>0</BurnRateExponent>
    <SolidGrainWts>True</SolidGrainWts>
  </PropellantData>
</Document>

reilleya, 2019-09-03:
Sorry about your propellants file! I hope it isn't too hard to reenter the info. I'd recommend keeping a version of the last release around for real use and keep `staging` for experimenting. That way you can always just copy over your propellants and config files from the stable release. The exhaust molar mass is found in the "molecular wt." field in ProPep.

benrussell11, 2019-09-03:
No problem as I should have backed up those files anyway.

benrussell11, 2019-09-03:
I’m getting the following warnings which I haven’t seen before:

 

1.	Constraint – Motor - Max pressure exceeded configured limit
2.	Constraint – Propellant - Chamber pressure deviated from propellant's entered ranges. Results may not be accurate.

 

#1 Preferences Max pressure is set to 3000 psi.

#2 Propellant chamber pressure range is set to 300 to 2000 psi

 

Any ideas as to why I’m seeing the above?

 

Ben

closed 2019-09-26 by reilleya