reilleya/openMotor #123: Have to run as an administrator on Windows

junwoo091400, 2019-09-06:
I downloaded the **V0.3.0 'packed' version**(one that's not 'unpacked') but when I opened openMotor.exe, I got the error **"Failed to execute script pyi_rth_qt4plugins".**

So I immediately suspected this as a some kind of a privilege issue, and when I tried _opening the program as an 'administrator'_, it successfully opened.

Is this an intended behavior? And can we just get rid of this administrator- related problem?

junwoo091400, 2019-09-06:
Seems to be a 'PyInstaller' related issues, might need to add some flags when compiling the code with PyInstaller, for now, I don't have the answer.

reilleya, 2019-09-07:
Hmm, I have not seen this problem on my own Windows install. Could I have some more details about your install? What version of Windows? Thanks for reporting this and looking into it. I'll try to get it fixed for v0.4.0.

junwoo091400, 2019-09-08:
I'm running 'Windows 10 Enterprise'. Hmm I really am not sure what is causing the problem, but according to[ some others](https://stackoverflow.com/questions/40716346/windows-pyinstaller-error-failed-to-execute-script-when-app-clicked), it would be helpful if I can see the trace / debug.

> Can you add "--debug" flag and also removing the "--windowed" when executing pyInstaller, and maybe I can execute the .exe file resulting from that, on my machine and get some insight into why this is happening.

If I have time, I will try to do that myself, but right now I don't have modules(like Qt4) installed.

reilleya, 2020-08-27:
Couldn't reproduce

closed 2020-08-27 by reilleya