reilleya/openMotor #147: Edit > Preferences > Units, Diameter not listed

dhbarr, 2020-04-24:
In Edit > Preferences > Units, I don't find the ability to change Diameter to mm ( most hobby motors are denoted this way ).

reilleya, 2020-04-24:
Changing the `length` setting changes the unit that is used for all fields with units of length, including diameter. If you change `length` to `mm` it will change every instance of `in` to `mm`. I'm not sure I want to have different units for different fields of the same base unit.

dhbarr, 2020-04-24:
It's a good point, I'm not a huge fan of mixed units either -- I notice that .eng and .rse files are both denoted in mm, so maybe just helping the user discover that to change the units for Diameter you need to change the units for Length?  And maybe all the places that's used?

Perhaps it's just me, I don't ever think "What's the length of this diameter?"

reilleya, 2020-08-27:
Not planning to do this, but I will add unit dropdowns to every field so you don't have to change settings to use a different unit.

closed 2020-08-27 by reilleya