reilleya/openMotor #149: Multi port nozzle

benrussell11, 2020-05-09:
Add support for multiport nozzles such as Aertech Medusa nozzle

reilleya, 2020-05-11:
This is a duplicate of #143 . I haven't had much time to work on this project lately but I plan to start again this summer once my data acquisition system is done and will think about how to do this then.

closed 2020-05-11 by reilleya

benrussell11, 2020-05-11:
Hi Andrew,

 

Ok … I’m planning on holding one of my intro classes at NYPower in July. I’ve updated the class to include the use of AT Experimental Motor Kits (EMKs) for those that already have the AT 54/1280 case. There are now two 54/1280 EMKs. One with the Medusa nozzle and one with a single port nozzle. So the need to support multi port nozzles has lessened to some extent.

 

Any chance you’ll be able to make NYPower now that URRF has been cancelled?

 

Ben

reilleya, 2020-05-13:
Sounds good, I'll put it on the backburner then. Yup, hoping to make it to NYPower as I was very excited for URRF.

benrussell11, 2020-05-13:
Cool ... See you at NYPower.

Ben