reilleya/openMotor #155: KNSU Characteristics

DiegoParedesE, 2020-06-14:
I saw a previous request to include KNSU in default propellants. Mod asked for characteristics. The following characteristics were sourced from Richard Nakka's Experimental Rocketry Web Site.

  | Parameter |   | Units | Note
-- | -- | -- | -- | --
  | Process method | Cast |   |  
Isp | Specific Impulse, ideal | 166 | sec. | 1
Isp | Specific Impulse, delivered   (typical) | 130 | sec. | 2
C* | Characteristic exhaust velocity,   theoretical | 3106 | ft/s |  
To | Combustion temperature,   theoretical @1000 psia | 1447 (1720) | deg Celsius (K) | 3
To | Combustion temperature,   measured@1000 psia | 1350 | deg Celsius | 4
ρ' | Density, ideal | 1.89 | gram/cu.cm. |  
ρ | Density, as cast (typical) | 1.80 | gram/cu.cm. | 5
ρ'/ ρ | Density ratio | 0.95 |   |  
X | Mass fraction of condensed-phase   products | 0.424 | - |  
k | Ratio of specific heats, 2-phase   flow | 1.044 | - | 6
k | Ratio of specific heats, static | 1.133 | - | 7
M | Effective molecular wt. of   exhaust products | 41.98 | g/mole | 8

Notes:
1.	At 1000 psi pressure; exit pressure one atmosphere
2.	Static tests AST-17, AST-18
3.	PROPEP combustion results
4.	Measured with type k thermocouple, static test AST-16
5.	Measured (liquid displacement method), typical grain for B-200 motor
6.	Effective (2-phase), at chamber conditions
7.	Value for exhaust mixture
8.	System mass divided by number of gas moles

**Burn rate behavior**
  | Burn rate behaviour | de St.Robert |   | Note
-- | -- | -- | -- | --
n | Burn rate pressure exponent,   strand | 0.319 |   |  
n | Burn rate pressure exponent,   erosive | 0.323 |   |  
a | Burnrate constant | 0.0665 |   | 1
ro | Burn rate @ 1 atm. | 0.156 | in/sec |  
r | Burn rate @ 1000 psia | 0.602 | in/sec |  
Tcr | Auto-ignition temperature | > 300 | deg. C. |  

1.	Use with pressure units of psi. Resulting burn rate is inches per second

**Chamber Pressure v.s. Kn** 
[image]

reilleya, 2020-06-14:
Great, looks like everything needed is here. Thanks!

reilleya, 2020-11-21:
[Done](https://github.com/reilleya/openMotor/commit/f0928b766223304e6d68a2eaaad5346874d25b98)

closed 2020-11-21 by reilleya