reilleya/openMotor #156: Bug - Preferences - Simulation Time Stamp

benrussell11, 2020-06-26:
Changing the Simulation Time Stamp in Preferences is not reflected in the export. Time Stamp interval is always .03 seconds. I changed it to 0.02. See the attached Preferences and csv.

Note - The preferences change is saved, just not used by export.

Ben
[image]
[EDW Sim #23.csv.txt](https://github.com/reilleya/openMotor/files/4837376/EDW.Sim.23.csv.txt)

reilleya, 2020-06-27:
This has confused a decent number of people so I should probably do something about it in the next release. The `Preferences` menu only changes the default values for this options that are used when a new motor is started. If you want to change them for a specific motor, you have to edit the `Config` option in the grain table:
[image]

benrussell11, 2020-06-27:
Ya that is a bit confusing. Good to know.

closed 2021-04-28 by reilleya