reilleya/openMotor #157: Info - Calculating exit pressure using nozzle expansion ratio

benrussell11, 2020-06-26:
Hi,

Can you please provide the formula for calculating the exit pressure using a nozzles's expansion ratio or point me to where I can find a reference?

BTW - Thanks for providing the exit pressure as one of the display options in OM.

Ben

benrussell11, 2020-06-27:
I decided to develop a spreadsheet to calculate thrust from an OM sim. I exported Time, Thrust, Pressure and Exit pressure for the attached motor. The propellant is MitchWhite. Exported pressure and exit pressure are in Pa. Nozzle dimensions are in meters. See attached spreadsheet, motor and exported csv.

I'm making the assumption that I should be able to use the exported data and Nakka formula as in the attached PDF, to calculate thrust and it should be equal to or very close to what OM calculates. Unfortunately it is not. My calculated thrust is lower.

In the spreadsheet, the cells labeled T1 - T4 are the calculations involving K as per the first page.

Can you please take a look at the spreadsheet and see if there is anything that I'm doing work? Or simply I can't do it this way and why.

What role does nozzle efficiency in the calculations?

Thanks,
Ben

[Nakkathr_example5.pdf](https://github.com/reilleya/openMotor/files/4841163/Nakkathr_example5.pdf)
[Pressure2Thrust.xlsx](https://github.com/reilleya/openMotor/files/4841164/Pressure2Thrust.xlsx)
[OMExample.ric.txt](https://github.com/reilleya/openMotor/files/4841168/OMExample.ric.txt)
[OMExample.csv.txt](https://github.com/reilleya/openMotor/files/4841175/OMExample.csv.txt)

benrussell11, 2020-06-28:
Ok I found the problem with the spreadsheet. T1 wasn't being calculated correctly - missing ()s. Updated spreadsheet attached. Calculated N vs OM N is within 3%. Wondering why I'm 3% high?
[Pressure2Thrust.xlsx](https://github.com/reilleya/openMotor/files/4842446/Pressure2Thrust.xlsx)

reilleya, 2020-06-28:
Glad you were able to figure it out! Nozzle efficiency is a factor multiplied onto the thrust coefficient to account for all of the performance losses that the program can't simulate, like the energy that is lost to heating up the case, liquid or solid exhaust products, and incomplete combustion in small motors. It has to be determined experimentally similar to characterization numbers like A & N, but once you get an idea of how well a propellant burns it becomes easy to guess. I usually use 0.9 - 0.95 depending on the propellant and the case. If you aren't accounting for efficiency your simulations will produce higher values. Also, openMotor takes the divergence angle and throat length into account when determining thrust coefficient, which Nakka does not.

benrussell11, 2020-06-29:
Thanks for the additional info. I'll take the nozzle efficiency % into account when I calculate the thrust coefficient. If there isn't already a request, can you add the thrust coefficient as an exported field.

I would like to be able to calculate nozzle exit pressure using the nozzle's expansion ratio. Where can I find the formula or references?

Ben

blackk100, 2020-07-16:
You need to use thermodynamic relations for isentropic flow.
This [Wikipedia article](https://en.wikipedia.org/wiki/Isentropic_nozzle_flow) should have them.

benrussell11, 2020-07-16:
Aryaman,

 

Thanks … I’m trying to solve for Mach in Excel which isn’t the easiest thing to do. Any help would be appreciated.

 

Ben Russell

ben.russell11@gmail.com

blackk100, 2020-07-17:
Hmm, I would recommend looking through Richard Nakka's SRM excel sheets as a reference, although most of his calculations are done with a combination of Excel Formulae and Macros: http://www.nakka-rocketry.net/softw.html#SRM

Both the 2014 and the v1 sheets are useful.

benrussell11, 2020-07-17:
I’ve been all over Nakka site and never thought to take a look at his spreadsheets. 

 

Thanks for the suggestion, I’ll definitely take a look.

 

Ben

closed 2020-08-27 by reilleya