reilleya/openMotor #166: Bug - App crashes after clicking on "Volume loading"

benrussell11, 2020-11-02:
Version ,5 as posted to Stage.

1. Open a sim file
2. Run sim
3. Click on "Volume load"
4 App cashes as per below

C:\openMotor>python main.py
QLayout: Attempting to add QLayout "" to MotorEditor "motorEditor", which already has a layout
C:\Users\ben\AppData\Local\Programs\Python\Python37-32\lib\site-packages\scipy\optimize\minpack.py:162: RuntimeWarning: The iteration is not making good progress, as measured by the
  improvement from the last ten iterations.
  warnings.warn(msg, RuntimeWarning)
C:\openMotor\motorlib\nozzle.py:47: RuntimeWarning: invalid value encountered in true_divide
  return fsolve(lambda x: (1/self.calcExpansion()) - eRatioFromPRatio(k, x / inputPressure), 0)[0]
C:\openMotor\motorlib\nozzle.py:47: RuntimeWarning: divide by zero encountered in true_divide
  return fsolve(lambda x: (1/self.calcExpansion()) - eRatioFromPRatio(k, x / inputPressure), 0)[0]
Traceback (most recent call last):
  File "C:\openMotor\uilib\widgets\resultsWidget.py", line 77, in xSelectionChanged
    self.ui.channelSelectorY.toggleEnabled(singleValueChannels, True)
  File "C:\openMotor\uilib\widgets\channelSelector.py", line 63, in toggleEnabled
    self.checks[channel].setEnabled(enabled)
KeyError: 'time'

Ben

reilleya, 2020-11-03:
Thanks for finding this, didn't intend for volume loading to be an option for the X axis (just the Y).

Hope all is well!

benrussell11, 2020-11-03:
Hi Andrew,

 

Working with my 2021 IREC team updating their test stand and code. Been interesting watching them do everything remotely as George Washington University is all remote classes. A couple of the students have apartments in DC. They had to get special permission to use the campus workshop.

 

All is well. I’m recovering from back surgery so things have slowed down a bit.

 

I pulled the latest copy of OpenMotor from Stage and have been playing with it. Thanks for add CF to the sim.

 

Ben

closed 2020-11-14 by reilleya