reilleya/openMotor #170: incorporate changes made in packaging for Debian

bdale, 2020-11-19:
I just packaged openMotor for Debian, with help from Petter Reinholdtsen.  In the process, we had to make a few minor tweaks to get things to run successfully outside a simple git cloned source tree.  I invite you to peruse what we did at https://salsa.debian.org/debian/openmotor and pull anything that seems useful.  See the debian/patches subdirectory in particular.

Going forward, it would be really helpful if all the sources could be pushed down into an openmotor directory so that it's easier to structure setup.py to install things in a cleaner manner.  We'd like to suggest you just do something like "mkdir openmotor; git mv main.py app.py motorlib uilib openmotor" and then adjust as necessary.

I look forward to contributing to openMotor beyond just packaging it for Debian over time...  thanks for providing such a cool bit of code under GPLv3!

reilleya, 2020-11-20:
Hey Bdale,

Thanks doing this! I'll take a look. I think I laid out the directories originally intending to rename `motorlib` and move it to another repo, but that hasn't happened yet. The directory rearrangement suggestion makes sense.

Let me know if you have any questions about the existing code or the issues tickets I've posted.

-Andrew

PS, I've been on a few group projects that used telemetrums and really liked them! I thought the open hardware model was neat, so I used it for my [motor test data acquisition system](https://github.com/reilleya/RMTS-Manual).

petterreinholdtsen, 2020-11-22:
Do you need a hand with the directory restructuring, or do you handle it yourself?

reilleya, 2020-11-22:
Hi Peter,

I haven't had time yet, but I plan to do it today or tomorrow.

-Andrew

petterreinholdtsen, 2020-11-30:
Very good.  I look forward to see the result.

petterreinholdtsen, 2020-12-03:
If you run into problems, please let me know and I can try to have a look.

reilleya, 2020-12-04:
Petter and Bdale,
Sorry for the delay! I'm pretty sure the changes in the `reorg` branch are what you have requested. You can now run the application with `python openMotor` when in the directory. Let me know if there's anything I've missed. I've done about two thirds of the tickets for the v0.5.0 release, which I'm hoping will be easier to package.
-Andrew

petterreinholdtsen, 2021-01-14:
I have not had time to test it.  Bdale, what about you?