reilleya/openMotor #173: MIME type for saved files?

petterreinholdtsen, 2020-11-30:
Hi.  For a GUI file browser on Linux to be able to open the saved files from openmotor, their MIME type need to be listed in the programs desktop file.  The mime type should be as specific as possible, and it is an advantage for correctly identifying the files as openmotor files that the magic database of the file(1) tool is able to identify openmotor files as files with this MIME type.

Is there a MIME type identified for use with openmotor?

reilleya, 2020-12-04:
No MIME type currently. The files have the extension `.ric` and contain YAML data. Is that specific enough to make one?

quigleyj97, 2020-12-04:
You can always just decide on an arbitrary one, MIME Types essentially boil down into fancy strings. You might want to sync with the maintainers of your Debian package to settle on a standard one, so they can implement it for Debian.

eg, application/vnd.openmotor

https://en.m.wikipedia.org/wiki/Media_type

petterreinholdtsen, 2020-12-04:
[Joe Quigley]
> You can always just decide on an arbitrary one, MIME Types essentially
> boil down into fancy strings. You might want to sync with the
> maintainers of your Debian package to settle on a standard one, [...]

It is also good to register the MIME type with IANA, see
<URL: https://www.iana.org/form/media-types >.

As for MIME type string, I agree that it make sense to put it under the
vnd tree.  I also suspect it make sense to add +yaml at the end to
indicate the underlying format, similar to the +xml, +json and +sqlite3
ones already registered with IANA.  My quick and dirty proposal would
thus be application/vnd.openmotor+yaml or text/vnd.openmotor+yaml.

Registering it early with IANA will avoid the confusion demonstrated in
<URL: https://stackoverflow.com/questions/332129/yaml-media-type > about
the YAML MIME type, and ensure all users of the software have a shared
view on how to handle OpenMotor file MIME types.

Is the charset in these files fixed to UTF-8, or can it vary depending
on the users environment?

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2021-01-14:
So, anyone willing to decide on a MIME type and stick to it as the official one?

petterreinholdtsen, 2021-02-01:
No one chimed in, so I guess no-one care.  I thus make random decisions to move this forward. I am implementing application/vnd.openmotor+yaml in the Debian package as the MIME type for OpenMotor .ric files, as a subtype of application/x-yaml.

petterreinholdtsen, 2021-02-01:
https://salsa.debian.org/debian/openmotor/-/commit/a60cc434c7d969782fccba00e4f6a7ae018d14a6 and https://salsa.debian.org/debian/openmotor/-/commit/e014c6cd2e9d09cf8d1b38c4b37603247c56a29b add a /usr/lib/mime/packages/openmotor and /usr/share/mime/packages/openmotor.xml to make sure Linux desktop file browsers understand that *.ric files should be handled by openmotor.

reilleya, 2021-02-02:
Thank you Petter! I haven't given this project the time it deserves in a while, but I'm happy progress is being made. I hope to finish v0.5.0 before too long.

petterreinholdtsen, 2021-02-02:
[Andrew Reilley]
> Thank you Petter! I haven't given this project the time it deserves in
> a while, but I'm happy progress is being made. I hope to finish v0.5.0
> before too long.

I am happy to be able to help.  But I do hope you will take the time to
register the MIME type with IANA, either the one I picked or some other
one.

I further hope you will documented the MIME type you pick in the README
and elsewhere, to make it more likely every user of the program end up
using the same MIME type.

-- 
Happy hacking
Petter Reinholdtsen

petterreinholdtsen, 2022-01-12:
Hi.  Did you have time to look at this the last year?

petterreinholdtsen, 2025-02-11:
And three years later, did anyone had time to decide on a recommended MIME type string in the mean time?