reilleya/openMotor #176: Location of "Releases" tab

bencoman, 2020-12-30:
The download instructions say...
> The easiest way to use openMotor is to navigate to the 'releases' tab above 

where I take to mean here...
[image]

and it was frustrating to not see a 'releases' tab there.  
Maybe github moved it?   Took me a while to find it on the right, so maybe will help others to adjust that description, and maybe include this pic...
[image]

bencoman, 2020-12-30:
Actually including a snapshot of the sidebar might seemed likely to confuse.

reilleya, 2021-01-02:
Thanks for taking care of this!

closed 2021-01-02 by reilleya