reilleya/openMotor #180: pyyaml update breaks invocation

bdale, 2021-01-31:
Apparently python3-yaml in Debian crossed the 5.1 threshold and now openmotor won't start.  There was a warning about yaml.load now needing a Loader= which I fixed (trivial patch below), but invocation still fails with:

bdale@rover:~$ openmotor
Traceback (most recent call last):
  File "/usr/bin/openmotor", line 33, in <module>
    sys.exit(load_entry_point('openMotor==0.4.0', 'gui_scripts', 'openmotor')())
  File "/usr/lib/python3/dist-packages/openmotor/main.py", line 15, in main
    app = App(sys.argv)
  File "/usr/lib/python3/dist-packages/openmotor/app.py", line 16, in __init__
    self.preferencesManager = uilib.preferencesManager.PreferencesManager()
  File "/usr/lib/python3/dist-packages/uilib/preferencesManager.py", line 47, in __init__
    self.loadPreferences()
  File "/usr/lib/python3/dist-packages/uilib/preferencesManager.py", line 56, in loadPreferences
    prefDict = loadFile(getConfigPath() + 'preferences.yaml', fileTypes.PREFERENCES)
  File "/usr/lib/python3/dist-packages/uilib/fileIO.py", line 37, in loadFile
    fileData = yaml.load(readLocation, Loader=yaml.FullLoader)
  File "/usr/lib/python3/dist-packages/yaml/__init__.py", line 114, in load
    return loader.get_single_data()
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 51, in get_single_data
    return self.construct_document(node)
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 60, in construct_document
    for dummy in generator:
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 413, in construct_yaml_map
    value = self.construct_mapping(node)
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 218, in construct_mapping
    return super().construct_mapping(node, deep=deep)
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 143, in construct_mapping
    value = self.construct_object(value_node, deep=deep)
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 100, in construct_object
    data = constructor(self, node)
  File "/usr/lib/python3/dist-packages/yaml/constructor.py", line 427, in construct_undefined
    raise ConstructorError(None, None,
yaml.constructor.ConstructorError: could not determine a constructor for the tag 'tag:yaml.org,2002:python/object/apply:uilib.fileIO.fileTypes'
  in "preferences.yaml", line 25, column 7

I tried removing ~/preferences.yaml and re-running.  The same error recurred, and it looks like the file was re-created with the same content?

In any case, this means I can't run openmotor at all, and I'm not adept enough with YAML to figure out what the right fix is?

bdale, 2021-01-31:
The yaml.load changes are just three instances of:

-        fileData = yaml.load(readLocation)
+        fileData = yaml.load(readLocation, Loader=yaml.FullLoader)

petterreinholdtsen, 2021-01-31:
By chance I discovered 'Loader=yaml.CLoader' seem to work better than FullLoader.

reilleya, 2021-01-31:
This is fixed in the latest staging version, using:
`fileData = yaml.load(readLocation, Loader=yaml.Loader)`

bdale, 2021-01-31:
Using yaml.Loader for the three instances works.  Thanks, guys!

closed 2021-10-10 by reilleya