reilleya/openMotor #181: headless broken in 0.4.0

bdale, 2021-01-31:
As I work on adding an autopkgtest for the Debian package, I'm having trouble making headless mode work.  I tried to just use one of the existing test motor definitions, and it fails like so:

bdale@rover:~$ openmotor -h debian/openmotor/test/data/real/o3100/motor.ric 
/usr/lib/python3/dist-packages/motorlib/nozzle.py:48: RuntimeWarning: invalid value encountered in true_divide
  return fsolve(lambda x: (1/self.calcExpansion()) - eRatioFromPRatio(k, x / inputPressure), 0)[0]
/usr/lib/python3/dist-packages/motorlib/nozzle.py:48: RuntimeWarning: divide by zero encountered in true_divide
  return fsolve(lambda x: (1/self.calcExpansion()) - eRatioFromPRatio(k, x / inputPressure), 0)[0]
/usr/lib/python3/dist-packages/motorlib/nozzle.py:13: RuntimeWarning: invalid value encountered in sqrt
  return (((k+1)/2)**(1/(k-1))) * (pRatio ** (1/k)) * ((((k+1)/(k-1))*(1-(pRatio**((k-1)/k))))**0.5)
/usr/lib/python3/dist-packages/scipy/optimize/minpack.py:175: RuntimeWarning: The iteration is not making good progress, as measured by the 
  improvement from the last ten iterations.
  warnings.warn(msg, RuntimeWarning)
Traceback (most recent call last):
  File "/usr/bin/openmotor", line 33, in <module>
    sys.exit(load_entry_point('openMotor==0.4.0', 'gui_scripts', 'openmotor')())
  File "/usr/lib/python3/dist-packages/openmotor/main.py", line 15, in main
    app = App(sys.argv)
  File "/usr/lib/python3/dist-packages/openmotor/app.py", line 48, in __init__
    print(motorlib.alertLevelNames[alert.level] + '(' + motorlib.alertTypeNames[alert.type] + ', ' + alert.location + '): ' + alert.description)

petterreinholdtsen, 2021-01-31:
This patch got the program limping along:

```
--- /tmp/app.py 2021-01-31 21:00:27.028684790 +0100
+++ /usr/lib/python3/dist-packages/openmotor/app.py     2021-01-31 20:59:09.468924398 +0100
@@ -3,6 +3,7 @@
 from PyQt5.QtWidgets import QApplication, QMessageBox
 
 import motorlib
+import motorlib.simResult
 from uilib import preferencesManager, propellantManager, simulationManager, fileManager, toolManager
 from uilib import importExportManager
 import uilib.widgets.mainWindow
@@ -45,7 +46,7 @@
                 motor = self.fileManager.getCurrentMotor()
                 simres = motor.runSimulation()
                 for alert in simres.alerts:
-                    print(motorlib.alertLevelNames[alert.level] + '(' + motorlib.alertTypeNames[alert.type] + ', ' + alert.location + '): ' + alert.description)
+                    print(motorlib.simResult.alertLevelNames[alert.level] + '(' + motorlib.simResult.alertTypeNames[alert.type] + ', ' + alert.location + '): ' + alert.description)
                 print()
                 if '-o' in args:
                     with open(args[args.index('-o') + 1], 'w') as outputFile:
```

I have no idea if it is the correct fix.

petterreinholdtsen, 2021-09-24:
Hi.  Any hope for a new release with a fix for this problem?

reilleya, 2021-09-25:
Hi Petter, with the chip shortage putting some of my other projects on hold, I have been planning to clean up the last few things for v0.5.0 and release it. This will definitely make it in when I do! I am working on finishing support for grains with conical cores, which is the last major feature planned for the release.

reilleya, 2021-10-03:
Should be fixed on `staging`, will make it into v0.5.0 soon.

closed 2021-10-03 by reilleya