reilleya/openMotor #187: TypeError: '<' not supported between instances of 'float' and 'NoneType'

devmalya-b, 2021-03-27:
For some specific configurations of the grain or the nozzle dimensions, the progress bar stops at some percentage. This is the traceback: 

**Traceback (most recent call last):**
  File "E:\DATA\Anaconda3_2020.11\lib\threading.py", line 932, in _bootstrap_inner
    self.run()
  File "E:\DATA\Anaconda3_2020.11\lib\threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "D:\D\Rocketry\Sims\Burnrate Sims\openMotor_0.4.0_Win\openMotor-staging\uilib\simulationManager.py", line 46, in _simThread
    simRes = self.motor.runSimulation(self.updateProgressBar)
  File "D:\D\Rocketry\Sims\Burnrate Sims\openMotor_0.4.0_Win\openMotor-staging\motorlib\motor.py", line 238, in runSimulation
    _, _, gamma, _, _ = self.propellant.getCombustionProperties(pressure)
  File "D:\D\Rocketry\Sims\Burnrate Sims\openMotor_0.4.0_Win\openMotor-staging\motorlib\propellant.py", line 49, in getCombustionProperties
    minPressure = float(minPressure)
TypeError: '<' not supported between instances of 'float' and 'NoneType'

This does not happen with 0.4.0 (or atleast not as often) but happens >50% of the times in 0.5.0 (after building the package).

reilleya, 2021-03-27:
Does this happen for a particular motor? Could you send an example file?

devmalya-b, 2021-03-28:
@reilleya this happens for specific dimensions of either the grains or the nozzle parameters. For example, I attaches a 2 grain (60mm in length) motor which stops at 56% (bug1.ric). If I change one of the grain's length to 62.5mm, the sim runs completely (notbug1.ric).

[openMotor_bugs.zip](https://github.com/reilleya/openMotor/files/6217153/openMotor_bugs.zip)

reilleya, 2021-03-29:
Good catch! This should be fixed in a push I just made. Let me know if you still have problems.

devmalya-b, 2021-03-29:
It works! Thank you.

closed 2021-03-30 by reilleya