reilleya/openMotor #188: Adding Thrust values in Motor Statistics

devmalya-b, 2021-03-29:
Could the values for Peak Thrust and Average Thrust be added to the Motor Statistics tab too? Kinda troublesome to try to approximate it from the graph.

reilleya, 2021-10-10:
Sorry for the late response! The average thrust is already included as the number in the "motor designation". It will always be in newtons regardless of your configured force unit because that's an amateur rocketry standard. I will try to figure out a spot to add a peak thrust output, but the statistics output is already getting pretty full. In the new v0.5.0 release, it is up to a 3x5 grid and I'm not sure I want to add another row.

reilleya, 2023-05-23:
No room in the UI, and easy enough to judge from the graph.

closed 2023-05-23 by reilleya