reilleya/openMotor #189: Burnsim vs OpenMotor

benrussell11, 2021-03-29:
Included are the Burnsim BSX file and the OpenMotor RIC file which was created via import from the Burnsim file.

Burnsim Version 3.0
OpenMotor 0.5.0 pulled from "Stage". Not the latest however.

Interesting to note the user has inhibited top of each grain.

The inhibited sim results are quite different.
 BS: M1501 TI:8630 NS
 OM: M117 TI:6705 NS

The un inhibited sim results
 BS: M1506 TI:8587 NS
 OM: M1119 TI:6715 NS

Curious as to why there is such a larger than normal difference between BS and OM?

Ben

[Rpt 2 98v1.bsx.txt](https://github.com/reilleya/openMotor/files/6222516/Rpt.2.98v1.bsx.txt)
[RRpt 2.ric.txt](https://github.com/reilleya/openMotor/files/6222517/RRpt.2.ric.txt)

reilleya, 2021-03-30:
Hi Ben,

It looks like both motors have the same value for "nozzle efficiency". In general, you will use a higher value in openMotor than you would in burnsim, because openMotor already accounts for some deviations from optimal thrust coefficient. For example, it takes into account the angle of the divergence cone and decreases the thrust coefficient accordingly, while burnsim assumes it is perfect and requires you to account for it in your own calculation/estimation of the efficiency. As the openMotor efficiency already accounts for most of the common sources of efficiency losses, I typically use a value of 0.95.

Additionally, it seems like the propellant that openMotor imported from that file has an ISP* of 213s. That is quite a bit higher than a typical ISP*, which are usually in the 160-190 range. Is it possible that you entered the delivered ISP from a motor into the ISP* field? That is also part of the difference, because oM works off of the gamma, molar mass, and combustion temperature it imports from the burnsim file rather than the ISP*. 

-Andrew

benrussell11, 2021-04-01:
Hi Andrew,

 

I met to get back to you sooner. That all makes sense.

 

The propellant and Burnsim are  from one of the IREC teams that I’m doing a safety review on. I like to do a comparison between OM and Burnsim as a double check. Usually there is that much of difference.

 

I’ve noticed most of the IREC teams that I’m reviewing, their ISP numbers have a tendency to be  high (>200) as you’ve noted. I always ask them how they got it and to clarify. A few do, most go silent.

 

Ben

closed 2021-04-27 by reilleya