reilleya/openMotor #191: Propellant editor input fields getting clipped on Mac

aushe02, 2021-05-09:
On Mac the propellant editor input fields are getting clipped.

Mac version - Catalina 10.15.7

[image]

JonahEmbry, 2021-12-10:
[image]
[image]
[image]

I'm seeing a lot of these clipped visuals

reilleya, 2024-12-05:
Better late than never! 

[image]
[image]

These problems were due to the UI library openMotor v0.5.0 used (Qt v5) not being able to handle display scaling very well, especially with the way the application used it. It worked fine at a scaling of 100%, but really struggled with anything else, making it not work on devices like the Microsoft Surface that have small, high resolution displays and use scaling by default.

 They finally have much better support in Qt v6, which when combined with some tweaks in the application code, results in the significantly improved results shown above. Still polishing things so it isn't merged yet, but it is much more usable on scaled displays now.

closed 2024-12-05 by reilleya