reilleya/openMotor #192: openMotor release v0.4.0 doesn't open on Mac

aushe02, 2021-05-09:
On Mac the latest release of openMotor does not open. 

Mac version - Catalina 10.15.7
openMotor version -  Release v0.4.0

Logs

May  8 20:24:24 com.apple.xpc.launchd[1] (com.invisionlabs.CraftManagerHelper[29610]): Service exited with abnormal code: 78
May  8 20:24:24  com.apple.xpc.launchd[1] (com.invisionlabs.CraftManagerHelper): Service only ran for 0 seconds. Pushing respawn out by 10 seconds.
May  8 20:24:24  com.apple.xpc.launchd[1] (com.apple.mdworker.shared.06000000-0200-0000-0000-000000000000[29569]): Service exited due to SIGKILL | sent by mds[165]
May  8 20:24:27  com.apple.xpc.launchd[1] (com.apple.mdworker.shared.07000000-0400-0000-0000-000000000000[29571]): Service exited due to SIGKILL | sent by mds[165]
May  8 20:24:32  com.apple.xpc.launchd[1] (openMotor.117892[29595]): Service exited with abnormal code: 255
May  8 20:24:34  com.apple.xpc.launchd[1] (com.invisionlabs.CraftManagerHelper[29617]): Could not find and/or execute program specified by service: 107: Malformed 

https://user-images.githubusercontent.com/32467683/117556824-034eff80-b03b-11eb-8cb6-9a4799e34587.mp4

reilleya, 2021-10-06:
Please try the release on [this page](https://github.com/reilleya/openMotor/releases/tag/v0.5.0-rc) when you get the chance and let me know if it still has problems. Thanks!

benrussell11, 2021-10-06:
Hi Andrew,

 

I was able to download the and install the Windows version w/o any problems. Did a real quick test and so far haven’t encountered any problems. More testing over the next couple of days.

 

When are you planning on releasing version .5?

 

Ben

closed 2021-10-10 by reilleya