reilleya/openMotor #194: Upgraded to stage. Sims for today as info. An a question

benrussell11, 2021-05-12:
Question - everytime I open a RIC file, it adds another entry for the same propellant because there is an existing propellant , but they have different properties. I believe they are different because of the precision being stored. I now have 7 entries for the "same propellant" in my list. Instead of automatically adding the propellant, suggest asking the user if they want to add or replace the current one. It would also be nice to know what is different.

FYI - Log from today. Yesterday, I updated from Stage. I did test the use of backspace and del keys. Will do some more testing as we have characterization static tests planned for the weekend.

Microsoft Windows [Version 10.0.19042.928]
(c) Microsoft Corporation. All rights reserved.

C:\Users\ben>cd \openmotor

C:\openMotor>python main.py
0.0000 [LOG] ################################################################################
0.0000 [LOG] Application started at 2021-05-08T13:14:31.842022
1.8910 [LOG] Starting new motor file
1.8910 [LOG] Opening window
QLayout: Attempting to add QLayout "" to MotorEditor "motorEditor", which already has a layout
2.4840 [LOG] Window opened
23.7030 [LOG] Doing a migration of a fileTypes.MOTOR from (0, 4, 0)
23.7030 [LOG]   Upgrading (0, 4, 0) to (0, 5, 0)
23.7030 [LOG] Saving propellants to "C:\Users\ben\AppData\Local\openMotor\openMotor/propellants.yaml"
23.7340 [LOG] The propellant from the loaded motor matches an existing item in the library, but they have different properties. The propellant from the motor has been added to the library as "Class TT ProPep (1) (4)"
29.2970 [LOG] Running simulation
C:\Users\ben\AppData\Local\Programs\Python\Python37-32\lib\site-packages\scipy\optimize\minpack.py:162: RuntimeWarning: The iteration is not making good progress, as measured by the
  improvement from the last ten iterations.
  warnings.warn(msg, RuntimeWarning)
C:\openMotor\motorlib\nozzle.py:47: RuntimeWarning: invalid value encountered in true_divide
  return fsolve(lambda x: (1/self.calcExpansion()) - eRatioFromPRatio(k, x / inputPressure), 0)[0]
C:\openMotor\motorlib\nozzle.py:47: RuntimeWarning: divide by zero encountered in true_divide
  return fsolve(lambda x: (1/self.calcExpansion()) - eRatioFromPRatio(k, x / inputPressure), 0)[0]
29.5160 [LOG] Simulation succeeded

C:\openMotor>python main.py
0.0000 [LOG] ################################################################################
0.0160 [LOG] Application started at 2021-05-12T10:37:32.094703
15.2350 [LOG] Starting new motor file
15.2660 [LOG] Opening window
QLayout: Attempting to add QLayout "" to MotorEditor "motorEditor", which already has a layout
16.8290 [LOG] Window opened
36.0790 [LOG] Doing a migration of a fileTypes.MOTOR from (0, 4, 0)
36.0790 [LOG]   Upgrading (0, 4, 0) to (0, 5, 0)
36.0790 [LOG] Saving propellants to "C:\Users\ben\AppData\Local\openMotor\openMotor/propellants.yaml"
36.1100 [LOG] The propellant from the loaded motor matches an existing item in the library, but they have different properties. The propellant from the motor has been added to the library as "Class TT ProPep (1) (6)"
44.6410 [LOG] Running simulation
C:\Users\ben\AppData\Local\Programs\Python\Python37-32\lib\site-packages\scipy\optimize\minpack.py:162: RuntimeWarning: The iteration is not making good progress, as measured by the
  improvement from the last ten iterations.
  warnings.warn(msg, RuntimeWarning)
C:\openMotor\motorlib\nozzle.py:47: RuntimeWarning: invalid value encountered in true_divide
  return fsolve(lambda x: (1/self.calcExpansion()) - eRatioFromPRatio(k, x / inputPressure), 0)[0]
C:\openMotor\motorlib\nozzle.py:47: RuntimeWarning: divide by zero encountered in true_divide
  return fsolve(lambda x: (1/self.calcExpansion()) - eRatioFromPRatio(k, x / inputPressure), 0)[0]
44.9220 [LOG] Simulation succeeded
66.1720 [LOG] Doing a migration of a fileTypes.MOTOR from (0, 4, 0)
66.1720 [LOG]   Upgrading (0, 4, 0) to (0, 5, 0)
66.1720 [LOG] Saving propellants to "C:\Users\ben\AppData\Local\openMotor\openMotor/propellants.yaml"
66.2040 [LOG] The propellant from the loaded motor matches an existing item in the library, but they have different properties. The propellant from the motor has been added to the library as "Class TT ProPep (1) (7)"
77.3440 [LOG] Running simulation
77.4220 [LOG] Simulation succeeded
89.8290 [LOG] Doing a migration of a fileTypes.MOTOR from (0, 4, 0)
89.8290 [LOG]   Upgrading (0, 4, 0) to (0, 5, 0)
89.8440 [LOG] Saving propellants to "C:\Users\ben\AppData\Local\openMotor\openMotor/propellants.yaml"
89.8750 [LOG] The propellant from the loaded motor matches an existing item in the library, but they have different properties. The propellant from the motor has been added to the library as "Class TT ProPep (1) (8)"
93.9540 [LOG] Running simulation
94.0320 [LOG] Simulation succeeded
123.4220 [LOG] Running simulation
123.5000 [LOG] Simulation succeeded
148.2350 [LOG] Running simulation
148.3600 [LOG] Simulation succeeded
159.4220 [LOG] Showing propellant menu
174.5000 [LOG] Propellant for current motor changed, updating all copies in history
177.5790 [LOG] Running simulation
177.6880 [LOG] Simulation succeeded
407.3750 [LOG] Running simulation
407.4850 [LOG] Simulation succeeded
478.6410 [LOG] Running simulation
478.7500 [LOG] Simulation succeeded
572.2820 [LOG] Running simulation
572.3910 [LOG] Simulation succeeded
643.0000 [LOG] Running simulation
643.1250 [LOG] Simulation succeeded
706.7820 [LOG] Running simulation
706.8750 [LOG] Simulation succeeded
770.1880 [LOG] Running simulation
770.2820 [LOG] Simulation succeeded
861.2500 [LOG] Running simulation
861.3290 [LOG] Simulation succeeded

C:\openMotor>

reilleya, 2021-10-10:
I think I know what is happening here. Say you have a propellant in your library and one in a file with the same name but different properties. The first time you load the motor, you'll get a message telling you that the properties differ and it'll create a new propellant with the properties from the file but with a number added onto the name. That's all correct and expected. The problem is that if you open that same file again, it is compared against the original propellant again and because it still doesn't match, a third propellant is added. This will repeat as many times as you open the file, which gets annoying as you mentioned. The temporary solution is to delete the original propellant and remove the `(1)` from the proper version of the propellant. Longer term, I will fix the code to make it so the propellant loader compares the version from the file against each separate instance of the propellant rather than just the first one. That way, if the properties match the `(2)` or `(3)` version, it won't make a new one.

reilleya, 2025-01-11:
Done on staging: [730fa05](https://github.com/reilleya/openMotor/commit/730fa05eb43ce665dcc8f1daf2a8b2741245a838)

closed 2025-01-11 by reilleya