reilleya/openMotor #199: OpenMotor 0.5.0 and Burnsim 3.3.x - Difference in total newton-seconds

benrussell11, 2021-12-30:
I created the attached motor design in OpenMotor. I then exported the OpenMotor ric to a Burnsim bsx file.

OpenMotor - M2515 Total Impulse = 8076.20 Ns
Burnsim - M2782 Total Impulse = 9040 Ns

My first thought was Burnsim isn't correctly taking the 1" end burning grain. Removed the end burning grain from both and reran the sims. There still is a large difference in the total impulse.

Any idea as to why the larger than normal difference in total impulse?

Ben

[98 V1 Test - Copy.ric.txt](https://github.com/reilleya/openMotor/files/7793680/98.V1.Test.-.Copy.ric.txt)
[98 V1 Test - Copy.bsx.txt](https://github.com/reilleya/openMotor/files/7793681/98.V1.Test.-.Copy.bsx.txt)

benrussell11, 2022-01-04:
If possible, I need an answer by the end of the week. as we are planning on a sub scale flight in 2 weeks using this motor.

Thanks,

Ben

reilleya, 2022-01-04:
Hi Ben,

I can look at the files once I'm back to my main computer later, but for now, the first thing that pops into my head is the efficiency field. If this value is directly copied between burnsim and openMotor you will see different results because burnsim does no additional thrust coefficient adjustment beyond multiplying by the efficiency, while openMotor takes divergence angle and throat length into account. First verify that the propellant mass is the same in each sim to make sure that all grains were copied over, then check efficiencies.

-Andrew

benrussell11, 2022-01-04:
Hi Andrew,

 

Thanks a for getting back to me. I really appreciate it.

 

I have checked and double checked as per below. The only difference I see is BS has a Nozzle Exit Dia of 2.55 and OM is 2.34 optimized. Efficiency is 85% for both. The number of grains is correct. I do have a 1.0” End Burn grain as the first grain. I deleted it from both sims, reran and still had a large difference in total NS.

 

What my IREC team has decided to do a couple of launches to verify their recovery and payload deployments. We want to keep the altitude between 5 and 6K feet. The grains are going in their 40” case with a spacer between the forward closure and forward snap ring.

 

Ben

reilleya, 2022-01-04:
The exit diameter being different will have an impact, as well as the efficiencies being the same as I mentioned in my last message. Because openMotor accounts for more than burnsim, you don't need to enter as much of a loss as you do in burnsim. Nearly all of my motors end up with ~0.95 efficiency in oM, while I was more used to 0.85-0.9 in BS.

benrussell11, 2022-01-05:
I bump the efficiency up to 95% and they are close. Let me know if you find anything else that would impact the total Ns.

 

Thanks,

 

Ben

closed 2024-12-05 by reilleya