reilleya/openMotor #200: Star grain - Convert from Burnsim to OpenMotor

benrussell11, 2022-01-11:
The included Burnsim file includes a Star grain which OpenMotor 0.5.0 can not import. The included OpenMotor design is my best guess for the OpenMotor Star grain.

How do I convert Burnsim's Major Width and Minor Width to OpenMotor Point length and Point base width?

Ben
[BS.bsx.txt](https://github.com/reilleya/openMotor/files/7847710/BS.bsx.txt)
[OM.ric.txt](https://github.com/reilleya/openMotor/files/7847712/OM.ric.txt)

AstroChuck, 2023-02-26:
I have written a converter for burnsim star grains that works on the included bsx file. Do you have any other burnsim star grain files I can use to test with?
[image]

AstroChuck, 2024-12-27:
#211 Addresses this capability. Please add additional star .bsx files if you'd like me to test them!