reilleya/openMotor #201: Propellant properties mistakes

NAOS-PUPPIS, 2022-05-03:
The data for the propellants DX and SB in the editor are not correct (burn rate coefficient) according to Nakka data

RandomDude4, 2022-08-25:
I agree, it seems to round to 0 when using metric units. Also KNSB and KNDX seems to have been swapped? But I am not sure I have not compared to actual Nakka data.

reilleya, 2023-01-14:
Interesting. That said, Nakka has multiple sets of data for many formulas, and I honestly think some of the piece-wise ones are just overfitting a small characterization run so I'm skeptical of his data. If anyone sends me a difference source for sugar propellant regression data I might switch over to that for the next oM version.

reilleya, 2025-03-15:
Fixed!

closed 2025-03-15 by reilleya