reilleya/openMotor #204: Automatic Motor Optimization

mvccogo, 2022-12-12:
It would be interesting to have an option in the "Tools" dropdown list to automatically find a suitable motor given the parameters proposed by the user. I've done a similar thing using scipy.optimize, and some of you may be familiar with Matlab's fminsearch/fmincon.

The parameters given as **constraints** are:
- Mean Pressure: average pressure developed after a simulation
- Port ratio: set minimum Port/Throat ratio to reduce effects of erosive burning
- Peak Pressure: peak pressure developed after a simulation
- Kn variation: maximum allowed Kn variation throughout the burn
- Maximum Mass Flux: reduce effects of erosive burning

The parameters given as **input/goals** are:

- Total impulse

The parameters given as **bounds** are:
- Min and Max outer diameter (BATES)
- Min and Max inner diameter (BATES)
- Min and Max length (BATES)

The **output** design parameters are:
- Number of grains
- Length of grains
- Inner and outer diameter of grains
- Nozzle configuration

Pseudocode/py code as a concept:
```py
def optimize_for_impulse():
    constraint = [{'type' : 'ineq', 'fun' : restraint_mean_pressure},              
                  {'type' : 'ineq', 'fun' : restraint_port_ratio},                    
                  {'type' : 'ineq', 'fun' : restraint_peak_pressure},               
                  {'type' : 'ineq', 'fun' : restraint_kn_variation},                  
                  {'type' : 'ineq', 'fun' : restraint_mass_flux}]                     
    bounds = ((0.005, 0.01), (0.05, 0.225), (0.003, 0.015))                    
    it = 0
    while (True):
        initial_guess = [random.uniform(0.005, 0.03664*0.6), random.uniform (0.1, 0.225), random.uniform(0.005, 0.015)]
        a = optimize.minimize(optimize_total_impulse, initial_guess, bounds=bounds, constraints=constraint ) 
        if a.success and a.fun < 10:
            return a.x, it
        elif it > 500:
            print("No suitable configuration was found in ", it, "iterations.")
            break
        it += 1
```

reilleya, 2022-12-15:
This would be fun to play around with! I still use openMotor fairly often but I don't have too much time to work on it. I would be able to find time to review a PR for this if you/someone sent one in, or will keep it in mind for when I someday start on a v0.6.0 if not.

mvccogo, 2022-12-15:
Sure, no problem. I'm also short on time, but I thought we should have an issue dedicated to this. I'll try building oM first, get familiar with the program then take a stab at it later on. Thanks, Andrew.

benrussell11, 2022-12-15:
I used OM all of the time. Spaceport America teams are also using it. This option would be great to have. I know I would use it.

 

For the 2023 cup, my team is working on designing a booster and sustainer for a 6” 2-stage with an on the pad weight in the 90 lb. range. Interesting to say the least.

 

Ben Russell

reilleya, 2024-12-05:
Hi @mvccogo! I'm cleaning up old tickets. Had time to play around with this?

mvccogo, 2024-12-06:
> Hi @mvccogo! I'm cleaning up old tickets. Had time to play around with this?

Hi! Unfortunately, I did not. It may be a good time to revisit the original proposal, after reading it again I got the impression I was way too focused on a total impulse as the only design goal, but there might be other goals like trying to match an input thrust curve (that would be awesome, as IIRC openMotor can handle generic grain profiles).

closed 2025-03-15 by reilleya