reilleya/openMotor #208: Ambient Temperature Dependence

1ckendall, 2023-01-18:
It's pretty common for solid propellants to have a degree of dependency on ambient temperature, so it would be really good to split the burn rate coefficient into a 'basic' term and a temperature dependency term when defining a new propellant. You could then set the ambient temperature in the simulation options, and perhaps a tool could run a set of simulations over a defined range of temperatures, so you could directly compare performance in different ambient conditions.

An example burn rate equation that I'm using is this:

$$
r = (2.29 \cdot 10^{-7} (T - T_{ref}) + 6.61 \cdot 10^{-5}) \cdot P^{0.305}
$$

With $T_{ref}$ set to 0 celsius.

At the moment I'm having to set a different propellant for every individual temperature I want to try (ranging from -15c to 30c), which is a bit cumbersome.

reilleya, 2023-01-19:
Huh, I was aware of the temperature dependance but had not thought to put it in the software. I always assumed it was negligible compared to other sources of variance for amateur motors and didn't worry about it. If you are working on a project where you have measured the difference I'd be very interested to hear about it if you have more you can share.

In general, I'm in favor of putting features in the software if:
1. There's literature to support the chosen implementation
2. There's a way for a typical amateur to measure/acquire the data required OR the feature is not required to perform a simulation
3. It doesn't clutter up the UI too much. There's a core set of features that a typical user needs and I don't want to bury them too much, especially with the display scaling struggles the UI still seems to have on some machines.

For 1 I'd love a link that shows a factor added on to A is a reasonable way to model this effect. Seems like 2 is all set because making this constant have a default of 0 would mean the average user who hasn't characterized at multiple temperatures could ignore the feature. 3 also seems ok if it is just an extra field in propellant editor, and a configuration value for ambient temperature. Nice idea!

1ckendall, 2023-01-19:
I've been working from the master's thesis of a previous member of my society ([https://repository.tudelft.nl/islandora/object/uuid%3Abd9fbf03-bf45-4bfe-aa27-39e1492de3e4?collection=education](url)), who I think probably chose that particular model based on this page by Nakka ([https://www.nakka-rocketry.net/bntest.html#Temperature](url)). I think that this model was chosen as a very simple way of expressing temperature dependence over the relatively restricted (vs the missile industry) temperature range amateurs face.

Alternatively, [https://doi.org/10.1080/00102207408960356](url) suggests

$$
r = r_0 \cdot e^{\beta(T-T_0)}
$$

With $r_0$ and $T_0$ values at a reference temperature, and $\beta$ the temperature variance coefficient. I think that this model may be more accurate, but perhaps harder for users to understand than the linear model.

I should be able to get some more data (for KNSB propellants, at least) in the next few months, that might indicate if the linear model or the exponential model is more appropriate for usage in the simulations? Unfortunately I'm still working on developing composite propellants, so I won't be able to test the models against APCP for probably at least a year.