reilleya/openMotor #210: Trying to update Star Grain Preview Widget causes a crash

AstroChuck, 2023-02-26:
Editing the geometry of a star grain causes a crash with the following error:
``0.0000 [LOG] Application started at 2023-02-25T17:09:27.724259
0.3590 [LOG] Starting new motor file
0.3590 [LOG] Opening window
QLayout: Attempting to add QLayout "" to MotorEditor "motorEditor", which already has a layout
0.4840 [LOG] Window opened
63.5630 [ERR] Traceback (most recent call last):
63.5630 [ERR]   File "C:\Users\drago\Documents\GitHub\openMotor\uilib\widgets\grainPreviewWidget.py", line 50, in updateView
63.5630 [ERR]     self.ui.tabRegression.cleanup()
63.5630 [ERR]   File "C:\Users\drago\Documents\GitHub\openMotor\uilib\widgets\grainPreviewGraph.py", line 41, in cleanup
63.5630 [ERR]     self.plot.lines.pop(0)
63.5630 [ERR] AttributeError: 'ArtistList' object has no attribute 'pop'
63.5630 [ERR]
63.5630 [LOG] Application closed due to exception.``

Just prior to the crash the file looks like this:
[image]

Attempting to edit the "Point Width" to 1.1 results in the crash. The value does not matter, as if 1.1 is entered initially, changing it to 1.0 will cause a crash.
This issue occurs both on the fix_tests branch and Release 0.5.0 on Windows (same computer), but not on Release 0.5.0 on Mac

An example of a file that can cause a crash is attached below.

[star_crash.ric.txt](https://github.com/reilleya/openMotor/files/10832593/star_crash.ric.txt)

AstroChuck, 2023-03-05:
I've done some more testing and this issue actually impacts all grain preview graphs any time the cleanup function is called (presumably when updating the preview widget).
This issue only occurs on my desktop computer, but crashes both OM 0.5.0 and my development environment. I suspect that my poor dependency hygiene is causing problems here and this is somehow tied to a new version of Matplotlib, although I can't confirm that yet. 
Using the method suggested by Matplotlib documentation for interacting with the artist list fixes this problem. Old code included in comments.
`            for _ in range(0, self.numContours):
                self.plot.lines[0].remove()
            #Old Cleanup For Loop
            #for _ in range(0, self.numContours):
                #self.plot.lines.pop(0)`

Flavsditz, 2023-06-10:
I am not sure if this is still stopping you, but if it is the fix for me was:
On the `gainPreviewGraph.py` on the `cleanup()` method I simply substituted this line:
```python
self.plot.lines.pop(0)
```
with this one:
```python
self.plot.lines[0].remove()
```

That being said probably fixing the versions in the `requirements.txt` is a good idea to have more control over these breaking changes

reilleya, 2023-06-11:
I have this fix and updated+fixed dependency versions on the `pyqt6` branch. Want to try it out? I'll merge it if it installs/runs properly for someone else too.

Flavsditz, 2023-06-11:
I have also pushed with my latest PR. Maybe if that one gets merged we are good on this front too.
But otherwise, do merge the pyqt6 branch and it is another step forward for the project :smiley:

closed 2023-12-09 by reilleya