reilleya/openMotor #214: PyQt6 for new contributions?

Flavsditz, 2023-05-04:
I have a mac with an m1 chip and I cannot get the project as is to compile.

Initially I cannot even install the pyqt5 dependencies since the project doesn't support the M1 chip. (See more here: https://stackoverflow.com/questions/70936664/metadata-generation-failed-while-installing-pyqt5)

Finally I also found out that PyQt6 does have support so I started to try and update the project for that only to see there is a branch that seemed to try this out.

My question is why is that branch not merged? Maybe why there seem to be so many stale branches. Some of them with really good ideas too!

I'd love to contribute with an idea to make the application translatable but for that I need to be able to test a bit more.
Would somebody be able to help me out? 

Otherwise I will dig out the old linux machine to get this going. :)

reilleya, 2023-05-05:
Hi! Cool to hear that you are interested in helping out. I specifically put together that PyQt6 branch in the hopes that it would fix some of the issues that the software currently has with [high DPI displays on Windows](https://github.com/reilleya/openMotor/issues/101), which were only mostly resolved in that ticket. Upgrading it didn't seem to help, but it sounds like there are other reasons to merge that branch. I can try to get it merged this weekend if you want to build off of it!

Flavsditz, 2023-05-05:
That would be really good!

On my experiments it got to a point that the `python setup.py build_ui` step failed due to the **PyQt Distutils** not supporting pyqt6. Did you manage to get the full workflow of building it working? Otherwise I would take a look at that too

Flavsditz, 2023-05-05:
@reilleya once you merged you can close this and I will know I can start cracking :) 
(I really appreciate it! Thanks again!)

reilleya, 2023-05-09:
> That would be really good!
> 
> On my experiments it got to a point that the `python setup.py build_ui` step failed due to the **PyQt Distutils** not supporting pyqt6. Did you manage to get the full workflow of building it working? Otherwise I would take a look at that too

Sorry for the delay! Did you get a specific error when you tried to build the UI? I was able to install the latest version of `pyqt-distutils` and it worked fine. I just force pushed the branch with a minor tweak to `pyuic.json` to remove something that was still pointing at a pyqt5 tool. Let me know if it is working better.

closed 2023-12-08 by reilleya