reilleya/openMotor #217: Altitude adjustment and nozzle size performance wrong?

csweeney0926, 2023-07-08:
Greetings,

I just started using open motor and I think I have identified a problem. I am modeling a particular motor and wanted to see what sort of adjustments would be needed to use the same motor as a 2nd or 3rd stage i.e. operating at a higher altitude/lower ambient pressure. I had previously used the optimize nozzle expansion to select the best exit diameter and was expecting to have to repeat that exercise at the higher elevation/lower pressure to find the new (presumable larger) optimum exit diameter. With a lower ambient pressure the nozzle optimized for ground level should have been under expanded (and therefore less efficient) higher up. However, open motor did not calculate a lower efficiency but rather a higher thrust. When the nozzle optimize function was run again at the higher elevation/lower pressure it did not come up with a different exit diameter.

The reason for the higher thrust at higher altitude is presumable the reduced friction from the ambient air on the exhaust plume because the air is thinner, which is nice feature and detail to have included in the calculation. However, the effect of the under expanded nozzle should be much greater than this and doesn't appear to be be included in the calculation. The nozzle optimizer is not capturing this and does not appear to take the ambient pressure into account when calculating the optimum exit diameter (which is a pretty significant issue).

It would be nice if the GUI of results presented some nozzle calculation data. The current version doesn't provide much detail on whether the nozzle is under or over expanded. The optimizer is nice though. Another feature that could be incorporated to further improve the nozzle accuracy is the motor speed and altitude vs time of the burn. With a known launch vehicle and a known motor one could, with a reasonable degree of accuracy, predict the elevation and velocity of the motor during the burn. Both variables would impact the performance of the motor and if they could be entered would improve the accuracy of the actual motor performance estimate.

reilleya, 2023-07-08:
Hi,
I assume from this bug report that you are using v0.5.0 build, like the windows installer or the mac app? This is a known issue in that version that was already fixed [here](https://github.com/reilleya/openMotor/commit/0d6c5a4d7fc72fde3b11b636689f0ec85c627c9c), so if you are able to run the application from source you can just check out the latest main to fix the issue. Otherwise, you'll have to wait for the next release, which I hope to do this year. I apologize for that being a vague timeline, but a release is a lot of work and I don't want to do one just to fix one bug. v0.6.0 will have more features too, so it'll be worth the wait.
-Andrew

csweeney0926, 2023-07-08:
Andrew,
Yes I am using v0.5.0 I did not know that this bug was already reported. So how do I implement the fix? do I just copy those corrected lines into the source code file? (Sorry for what is probably a dumb question, I am more of a hardware person).
Colin

reilleya, 2023-07-08:
If you are running from source (running the application using `python`), you can just download/clone/pull the latest `staging` and it'll be fixed. As I mentioned in the last post, if you are using the windows installer version or the mac .app, it isn't as easy to patch, and you'll have to wait for v0.6.0 to come out.

csweeney0926, 2023-07-08:
Yeah I am running the windows installer, so I guess I'll wait for the next update. Thanks for responding by the way
Colin

reilleya, 2023-07-08:
Of course! Let me know if you run into other issues or have ideas for improvements.

closed 2023-07-08 by reilleya

csweeney0926, 2023-07-10:
Will do
Colin