reilleya/openMotor #223: Endburning grain simulation doesn't work

Arcturusastro, 2024-01-22:
Hi,
When trying to simulate an end burning grain, the simulation only lasts 1 time step and doesn't output anything useful. Other grain geometries work just fine. The chamber pressure and Kn is nonzero for that one time step, but thrust is zero. Looks like it has to do with web burnout detection since it's already pretty much 0 after that one time step.

reilleya, 2024-01-22:
Hi! Do you have an example motor file that shows the problem?

Arcturusastro, 2024-01-24:
Github won't allow me to attach the .ric file since it isn't supported. Here's a google drive link. https://drive.google.com/file/d/1RTZ-28XAUquSxvzftEWoTYfh8k1tP_ya/view?usp=sharing

Arcturusastro, 2024-01-30:
One thing I noticed while testing propellant characteristics with the endburning grains - If the burn rate coefficient is ridiculously high, the simulation runs as expected, however I can't get it past the first time step with a normal propellant.

reilleya, 2024-01-31:
Finally got the chance to look at this! There's two issues with the design, either of which is enough to cause the confusing behavior you are seeing in the simulator. First, the Kn is far too low to get a good burn out of a formula like CL. It requires a Kn in the 200-400 range to burn at a high enough pressure to regress at a reasonable rate, and your file has a Kn of 64. You'll find that the nozzle throat for this motor has to be closer to 0.2-0.25" to get to a Kn in this range. As an aside, a moderate burn rate propellant like CL isn't well suited as an endburning formula, because even at a Kn of 400, the grain in the file you sent will take 2.5 minutes to burn!

The other issue is the expansion ratio. The exit diameter is way too large for really any motor with a throat of that size, and _especially_ one with as low of a chamber pressure as this one. The expansion ratio is so high that the motor generates effectively 0 thrust, which causes the simulation to end in one timestamp. At ambient pressure and typical amateur motor chamber pressures (300-1200 psi), the optimal expansion ratio is likely in the 4-10 range, and much outside of that range will produce poor results.

The software should absolutely make this more clear with a warning of some sort, I'll play around with adding one to help people avoid getting confused about this. I'll probably add a warning for simulations that end with a substantial amount of propellant left, indicating that the motor didn't generate thrust.

closed 2024-01-31 by reilleya