reilleya/openMotor #224: Propellant preset disappeared

Lordakos, 2024-01-26:
I have mistakenly deleted a propellant from the editor, one that came with openmotor. I have uninstalled the program several times, downloaded and installed it again, but I cannot get the propellant back. Is there a way I can get it back somehow? Thanks!

mechandansinha, 2024-10-26:
I have exactly the same issue and looking for the solution frantically. I don't understand why uninstalling and reinstalling the software doesn't restore the default options. Weird! Also, I don't see any solution posted here and it's been 9 months, so pretty disheartening. Someone please help.

reilleya, 2024-10-26:
Hey, sorry for the delay! Yeah, the uninstaller probably should clean up the propellant library, or at least offer you the option to. If you want to reset it back to the defaults, just delete the `propellants.yaml` file in the data directory, which is `C:\Users\andrew\AppData\Local\openMotor\openMotor` for me on Windows 11. When the program starts and notices the file isn't there, it'll create it. 

If you want to keep any custom propellants you entered and get the defaults back as well, don't delete the file, just rename it. Start the program to generate the default file again, and then open both in a text editor and copy your propellants in to the `propellants.yaml` file

mechandansinha, 2024-10-26:
@reilleya Thank you sooooo much!!! Yes, this solved the problem.

closed 2024-12-05 by reilleya